OctoPrint/OctoPrint-MQTT #1: Installed, No error messages, but no messages arrive on broker

ukmoose, 2015-05-05:
Hi, I love the idea of this to enable me to integrate messages from Octoprint into my Node-Red messaging system.  However after having setup the plugin nothing is received by my MQTT broker.  I'm sure I'm probably doing something wrong but I can't fathom what!

1. What were you doing? 
Migrated to Devel branch of Octoprint from Octopi.
Installed MQTT plugin using following command:
 /home/pi/oprint/bin/pip install https://github.com/OctoPrint/OctoPrint-MQTT/archive/master.zip
Added Mosquitto broker details into /home/pi/.octoprint/config.yaml
Copied mqtt_test.py text into file in ~/.octoprint/plugins
Restarted Octoprint

2. What did you expect to happen?
Expected to start receiving MQTT messages on broker ( broker is receiving other messages within system)
3. What happened instead?
No messages appear to have been sent

4. Branch & Commit or Version of OctoPrint:
Version: 1.2.0-dev-710-g88bc0b7 (devel branch)

(and of Mosquitto)
Mosquitto v 1.3.4 running on Mac OS X 10.10.3

5. Printer model & used firmware incl. version
  Printrbot Simple Metal
6. Browser and Version of Browser, Operating 
   Safari Version 8.0.4 (10600.4.10.7) Mac OS X 10.10.2

7. Link to octoprint.log on gist.github.com or pastebin.com
https://gist.github.com/ukmoose/6d61689d178292eb1198

8. Link to contents of terminal tab or serial.log on 
  serial log file size is 0 bytes

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

N/A

I have read the FAQ.

foosel, 2015-05-07:
That's just a shot in the dark for now (I'm currently on a business trip and can't test it myself) but try updating your OctoPrint version. I did some heavy development over the past weeks (+120 commits on top of your version) and among those was introducing a plugin lifecycle management, which came *after* your OctoPrint version, but which this plugin was already modified for and depends on.

Sorry, you are really running the bleeding edge here :)

ukmoose, 2015-05-08:
I recognised I was bleeding edge, but the lure of MQTT was too strong :-)

I updated OctoPrint and I still wasn't receiving any MQTT messages.
So following your comment on plugin lifecycle management I installed the plugin manager (the next step was to start again so I didn't have much to lose!)
After installing the plugin manager I have now started receiving MQTT messages.  

Many thanks for all your efforts 

Mark

closed 2015-05-08 by ukmoose

foosel, 2015-05-08:
Hm... a bit surprising though that it needed the plugin manager (it only makes installing and enabling/disabling plugins easier and shouldn't influence the basic operation of the plugin system). Odd. Ah well... I'm going to make the plugin manager part of OctoPrint anyhow soon, so let's hope it was just a hiccup :)