OctoPrint/OctoPrint-MQTT #2: version compatible?

brunkj, 2015-12-31:
does this work with 1.2.7 and 1.2.8?

foosel, 2016-03-06:
Sorry, the notification for this must have gotten buried in all the other OctoPrint related github messages, I only just now saw it :(

I guess you are talking about OctoPrint? It definitely should.

brunkj, 2016-03-06:
hmmm haven't been able to get it to work.

foosel, 2016-03-06:
Some more details would be nice ;) Were there problems installing it? Configuring it? Running it? What happens? What did you expect to happen? What didn't, what did, what did you try? What's in `octoprint.log`?

I don't sit in front of your setup, I can't see the issue myself, so I can't help you unless you tell me about it.

Locbit, 2016-03-25:
I have a question can I use this plugin to post basic JSON.

foosel, 2016-08-29:
@Locbit sorry, I don't understand the question, please provide more details (in a different ticket). It's a plugin for getting OctoPrint to talk to MQTT brokers.

@brunkj anything more to provide here? If not, I'll close it.

closed 2017-02-15 by foosel