OctoPrint/OctoPrint-MQTT #5: error: [Errno 111] Connection refused

s00500, 2016-04-07:
Mqtt is not working and when I start octoprint manually I found this strange error

I added some debugstatements in the plugin... the _on_connect callback never gets called because of the error in socket.py... why does this happen?

2016-04-07 21:16:42,574 - octoprint.plugins.mqtt - INFO - before conn async
2016-04-07 21:16:42,576 - octoprint.plugins.mqtt - INFO - connect async
2016-04-07 21:16:42,578 - octoprint.plugins.mqtt - INFO - loop started
Exception in thread Thread-11:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 505, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/paho/mqtt/client.py", line 2283, in _thread_main
    self.reconnect()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/paho/mqtt/client.py", line 734, in reconnect
    sock = socket.create_connection((self._host, self._port), source_address=(self._bind_address, 0))
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err
error: [Errno 111] Connection refused

jawilson, 2016-08-27:
Can you post your plugin configuration? What kind of MQTT broker are you running?

s00500, 2016-08-28:
Oh sry, it was an issue with mz brokers firewall, sry that I didn't close this issue

closed 2016-08-28 by s00500