OctoPrint/OctoPrint-MQTT #10: Simple message and callback

zaapp, 2017-03-21:
I would like to connect OctoPrintto my home monitor on node-red.
How can i check status of various events provided by OctoPrint via MQTT?

Example:
Send command to read that any client is connected to Octoprint webpage.
Octoprint reads message and publish status of event.

ukmoose, 2017-03-21:
What issue are you having? Use this plugin. Configure it with your MQTT topic and MQTT broker details.   
Point a MQTT IN node in node red at your broker and subscribe to the topic you configured. It works

zaapp, 2017-03-21:
But question was about MQTT OUT node to check status. Messages published by Octoprint works well :) but when i want to check status manually i can't do that or i do something wrong.

foosel, 2017-03-22:
I think you are not understanding how MQTT works, or I don't understand your question at all.

OctoPrint publishes a message on your broker. Whatever is subscribed to your broker on that message gets that message and can act on it. That's it. You'll probably want your nodered install to subscribe to the topics that OctoPrint publishes messages on and take it from there.

The other direction (something else publishes a message on some topic on your broker, OctoPrint receives that message and acts on it) is in principle possible - the code is there - but there isn't anything actually using that code right now in that plugin because I don't feel comfortable doing that considering potential security implications. There is a documented API however for plugins to further extend OctoPrint using the MQTT functionality provided by this plugin.

closed 2017-03-22 by foosel