OctoPrint/OctoPrint-MQTT #14: Heatup-information broadcast?

Xasin, 2017-07-07:
Hi! 
This is more of a suggestion than an issue, but I was wondering if it would be possible to send out the printer's temperature during heat up-phases? 
Maybe a hook into Octoprint's temperature readouts could be made, and the printer temperature, as well as status (like heating, heated, cooldown/off), could be sent out? 

It would be cool to have an IoT-Connected display device show a small bar graph of the current temps of the printer as it warms up!

rperciaccante, 2017-07-10:
+1 on temp in status messages - both in heat up and during printing, something like this maybe?

/printer/progress  {"target_bed_temp": 60, "target_tip_temp": 220, "actual_bed_temp": 54.6, "actual_tip_temp"; 214}

Being able to put a small LCD display, or be able to incorporate the device into existing monitoring environments would be awesome!

foosel, 2017-07-10:
Would be possible, I do lack the time however to do this sooner rather than later.

If anyone wants to tackle this, the plugin would need to register itself as a [PrinterCallback](http://docs.octoprint.org/en/master/modules/printer.html#octoprint.printer.PrinterCallback) with `self._printer` in order to get temperature updates pushed - we don't want to have to constantly query stuff, getting it pushed has way less of an impact performancewise.

Posting the current temperatures to a topic, with the internal data structure ("{'tool0': {'actual': ..., 'target': ...}, 'tool1': {'actual': ..., 'target': ...}, 'bed': {'actual': ..., 'target': ...}}") should be achievable that way.

Xasin, 2017-07-10:
Hmrpf ... 
Python, I'd have to work my way into that before I could really do something ... But it looks fairly similar to some Ruby structure.

If no-one else grabs it by the end of the week I'll try out if I can get it working on a fork.
Would be cool though to have my [RGB Circle Display](https://drive.google.com/open?id=0B11Wi349XkO7ZTItR2piUFE4VHc) able to show me how my printer's temperature is doing.

rperciaccante, 2017-07-10:
I am just about useless with python at any appreciable speed, but I will help if I can...  i can throw a few $$ in on a bounty as well...

derpicknicker1, 2017-07-10:
See #15 :grinning:

foosel, 2017-07-12:
Merged the PR, will be part of the next release (0.5).

Xasin, 2017-07-12:
Alright, looks like all necessary changes are in place. I suppose the issue will be closed automatically once the PR gets merged ^^

foosel, 2017-07-14:
Not fully automatically it seems. Hence me now closing it manually because I just tagged and released 0.5 including this.

closed 2017-07-14 by foosel