OctoPrint/OctoPrint-MQTT #16: Feature request: Time remaining

liftoff, 2017-07-20:
Doesn't appear that the plugin sends out the time remaining on any channels.  Is there any way we can get that info published?

sprior, 2017-09-11:
I'd love to see time remaining/time elapsed added as well.  MQTT is great, I've already added an Octoprint Progress gauge to a node-red dashboard.

kanflo, 2017-12-13:
That is something I would like too. It does however require a change to OctoPrint rather than this plugin as OctoPrint currently does not send the time remaining to the plugins.

kezmdor, 2017-12-14:
I am also very interested in something like ETA or ETL.

dp106, 2018-01-08:
You could query the api to get information on the current job and time remaining:  
[http://docs.octoprint.org/en/master/api/job.html#id3](url)
(You might need to scroll down a bit to find the section "Retrieve information about the current job")

jneilliii, 2018-10-12:
It looks like this may have been accounted for in the devel branch, but has yet to be release to the official version.  I'm going to test the devel branch over the weekend and if everything looks stable will push to master and release official version 0.8.0 next week.  @tedder & @Kunsi, is that good for you guys?

tedder, 2018-10-12:
Oh, nice. I was going to explore what data was in the Octoprint hook to see if we could support it. That's better yet.

jneilliii, 2018-10-13:
I just made a small change to dev to only include the extended printer_data attribute to the printing progress topic rather than with every topic.  That seems to make more sense to me, but let me know what you guys think.

jneilliii, 2018-11-09:
This feature has been released as version 0.8.0.  Ready for your fixes @tedder on the other issues.

PR #41 (2018-05-09): Option to include all printer data in events
This fixes #16 , as well as providing all of the available information in each MQTT message (as an option).

(With no PEP8)

Thanks!
closed 2018-11-09 by jneilliii