OctoPrint/OctoPrint-MQTT #17: Any way to detect M600 (filament change) and publish that?

liftoff, 2017-07-20:
I tested some gcode that incorporates a filament change (M600) and watched the output of the plugin on `octoprint/#` but I didn't see anything that would indicate a filament change status.  Is this a feature that can be added or is it something I would need to implement via a plugin?

Thanks for your assistance with this.  My printer is in my garage and I can't hear the beeps when it's time for a filament change.  Writing something to notify me when a specific MQTT message is published would be an easy fix to that problem!

DominikPalo, 2019-05-12:
Any updates? It would be really helpful also for me :)

jonferreira, 2019-09-11:
any updates? would really want to see this implemented

jneilliii, 2019-09-28:
I was just looking at this and was wondering what happens on the OctoPrint side when a M600 command is sent currently?  In theory shouldn't it pause the interface until resumed?  If that is the case couldn't you subscribe to the topic `octoprint/event/PrintPaused` and react to that? I know that's not a specific M600 event, but the gcode scripts that are interpreted by the MQTT plugin are the ones listed [here](https://docs.octoprint.org/en/master/events/index.html#gcode-processing). If M600 doesn't pause the print then you could actually add that by adding M600 to the serial connection settings on the firmware & protocol tab.

[image]

jneilliii, 2019-09-29:
Another option would be to add an M300 just before the filament change in your gcode and react to the topic `octoprint/event/Alert`.

AJErazzor, 2020-08-17:
> I was just looking at this and was wondering what happens on the OctoPrint side when a M600 command is sent currently? In theory shouldn't it pause the interface until resumed? If that is the case [...]

This worked brilliantly for me until 1.4.2 (I think) where the M600 pause command which was there by default was removed.  When adding in again - the behaviour is now - printer does filament change process, MQTT detects pause, however the GUI does not show this and when the filament change is complete on printer, it used to just start again but now once the filament change is done, the GUI THEN changes to pause and I need to resume it from the GUI - is there another command we need in there instead of m600 now?  Printer is Prusa Mini

jneilliii, 2020-08-18:
That sounds like a breaking change in OctoPrint, or more specifically the Action Command Plugin that is bundled with OctoPriint 1.4.2. I haven't looked to see if there are any changes there, but I know there were some additions like the notifications panel in the sidebar, etc. Maybe post the question to OctoPrint issue tracker directly.