OctoPrint/OctoPrint-MQTT #20: Feature Request: MQTT Subsicribe to Issue Commands to Octoprint

HotTabascoSauce, 2017-11-01:
It would be cool if you could issue MQTT publications on another device and have the octoprint perform actions based on subscriptions. Either octoprint native actions or sending Gcode options to the printer.

For example, if you were to send {"message": "connect", "port": "/dev/ttyUSB0", "baudrate": "115200"} to the topic /octopi/cmnd/ then the octopi instance would attempt to connect to the printer on USB0 at the given baud rate.

Or {"message": "position", "home": "all"} to the topic /octopi/cmnd/ then octopi would sent G28 to the printer while {"message": "position", "home": "X Y"} would home just the X and Y axis with G28 X Y.

I guess you might not need an entire library for gcode, could do something like {"message": "printer control", "gcode": "G28 X Y"}

Incoming messages from other devices could be executed immediately with a "now", or if printing, queued and executed when printer status is Operational

nightcat91, 2017-12-30:
I would also like some sort of control over MQTT.
Just beeing able to use MQTT to change the temperatures would be a great help!

dp106, 2018-01-08:
Ah, I've just created a new topic asking for the same thing! :)

Only solution I've found so far is to use the api to issue a command.

derpicknicker1, 2018-01-12:
The plugin has helpers to subscribe and publish to topics. This can be used by other plugins. So you are welcome to write your own plugin to handle commands via mqtt.

dp106, 2018-01-13:
Not sure I follow, I don't want to write another plugin. Rather as this is the MQTT plugin I would like it to be able to accept incoming messages as well as publishing them. 

I'm probably using completely the wrong terminology as have been using MATT nodered etc for about 3 days 😀

SoundGuy, 2018-09-11:
I would love to be able to control my printer through home automation (Alexa/siri) and use MQTT commands to send GCode commands for that.
Example to tell it to do stuff like "Alexa, pre-heat PLA" . 
This would be really cool do control my printer handsfree. I can already turn it on and off with voice commands using Sonoff.

rpitera, 2018-10-03:
Add me to the list as someone who'd really love an 'in' topic for commands!

SoundGuy, 2018-10-10:
how come MQTTSubscribe didn't appread in the plug-in manager, and i had to install it manually through the URL you provided?

jneilliii, 2018-10-10:
Because I haven't released it to the plugin repo as I haven't thoroughly tested it.

SoundGuy, 2018-10-15:
I managed to get it working, sort of. Posted in the other project. 
I'll make a video of me turning on the printer with Siri soon.

hakan42, 2018-10-15:
> I managed to get it working, sort of. Posted in the other project.

Which "other project"? I'm totally curious how you did it.