OctoPrint/OctoPrint-MQTT #24: Feedbak when printing from SDCARD

Allycat1972, 2018-01-03:
Hi. Printing using the Octoprint server is sometimes very slow when printing complex shapes so I often print from the SDCARD. Currrently there is no status / update when using this method.  Is there any way to get regular status / feedback from the 3D Printer via MQTT?  I publish the MQTT topics using OpenHAB so I can monitor the progress remotely. Many thanks, Andrew.

foosel, 2018-02-21:
I cannot reproduce this here, progress reports while printing from SD arrive just flawlessly.

Of course, the SD print must be started *through OctoPrint*, otherwise OctoPrint won't even know you are printing from SD and also can't check progress with your printer.

foosel, 2018-03-23:
@Allycat1972 Can this be closed?

Allycat1972, 2018-03-23:
@foosel Yes, thank-you for the response.

closed 2018-03-23 by foosel