OctoPrint/OctoPrint-MQTT #25: Feature Request: Add Disconnected Topic

dp106, 2018-01-08:
Firstly would just like to say how good this is, very impressive. I've been playing around with this getting it set up all weekend :)

Currently you have the connected topic:  "octoprint/event/Connected" which is broadcast when OctoPrint connects to the printer but there is no corresponding disconnected topic.  The only way I can see to do this is by making a call to the api to get the current state of the connection.

hovissimo, 2018-03-29:
The MQTT way of doing this is to set a "last will and testament" (LWT) message with the broker.  The broker then publishes the message as specified when the broker detects the client is no longer connected.

Random google link: https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament

This is actually mentioned in the readme as a TODO.

foosel, 2018-04-06:
I just realized that I read this wrong and you are actually talking about the printer connection here, not the MQTT connection. Well, 0.7 will have LWT support anyhow ;)

However what you are asking for here should actually already work and in my test also does, see the highlighted message here:

[image]

The events aren't retained though (that wouldn't make a lot of sense considering that events are not a status), so you *must* be connected to the broker and subscribed to those messages to actually receive them. Also note that the list of events in the README isn't exhaustive, for a full list of all available events consult [OctoPrint's docs](http://docs.octoprint.org/en/master/events/index.html#available-events).

I honestly don't want to mirror each and every status available on the REST API, if you need that kind of control I suggest to use that API instead, or alternatively implement another plugin utilizing the helper functions provided by this one.

closed 2018-04-06 by foosel