OctoPrint/OctoPrint-MQTT #33: [Bug]? /event/ UpdatedFiles being updated after last print completes.

RJ-Make, 2018-02-25:
#### What were you doing?

My home automation system (HomeSeer) watches this topic /event/ UpdatedFiles and if a message is received, the HA will provide power to the printer after 2 min. 

With the latest version (0.6), a message is being sent on /event/ UpdatedFiles at the end of a print, (without a file being updated). 

This of course is causing the printer to restart.

#### What did you expect to happen?

Only have a message sent when an actual file update takes place. 

#### What happened instead?

A message is being sent after a print completes, when no file updates take place. 

#### Version of OctoPrint

1.3.6

#### Version of the MQTT plugin

0.6

foosel, 2018-02-26:
I can't reproduce this here. 

In general the plugin doesn't generate any events on its own, it just forwards whatever it gets from the underlying OctoPrint install. Try disabling all other 3rd party plugins but the MQTT plugin, see if the issue reproduces then. If it doesn't, some other plugin you have installed things that cause an additional `UpdatedFiles` event.

Based on the description of your setup, I'd actually also not trigger on `UpdatedFiles` but rather on `FileAdded`. `UpdatedFiles` gets triggered on every file list changes, including removes and folder management, and it doesn't sound like powering up your printer just because you are cleaning up your file list would make a ton of sense.

RJ-Make, 2018-02-26:
Thanks Gina, Good advice. I'll run some test this week, and switch over the logic in my HA code.

foosel, 2018-03-23:
@RJ-Make can this be closed? If so, please do it ;)

foosel, 2018-04-06:
Closing since no reply.

closed 2018-04-06 by foosel