OctoPrint/OctoPrint-MQTT #34: mqtt does not connect to host

paulkoan, 2018-03-23:
#### What were you doing?

I have installed octoprint-mqtt on octopi with the plugin manager.  The installation look good, I can configure and save settings.  My octopi server never connects to my mqtt server.

Ideally provide exact steps to follow in order to reproduce your problem: -->

1. Install octoprint-mqtt
2. Configure host and port in mqtt settings, enable event flags for everything and activate (tried both IP and hostname)
3. Reboot
4. Start print, change settings, anything that might cause a mqtt message

#### What did you expect to happen?

I expected to see activity on /octoprint on my mqtt server.  I also expected to see traffic in tcpdump to my mqtt server IP  on port 1883 from the octopi (I can see pings).  Both devices are on the same network and can interact otherwise vi ping and `nc <ip> 1883` works from the octopi machine,

#### What happened instead?

No connection attempt is made to my mqtt server

#### Version of OctoPrint

     OctoPrint 1.3.6 running on OctoPi 0.14.0 

#### Version of the MQTT plugin

MQTT (0.6)

#### Used MQTT broker and its version
mosquitto version 1.4.14 (build date Mon, 10 Jul 2017 23:48:43 +0100)

mosquitto is an MQTT v3.1.1/v3.1 broker.

#### Link to octoprint.log

Will add

foosel, 2018-03-23:
Please give me a ping once you've added the log.

paulkoan, 2018-03-23:
Sure - I didn't add it immediately as I saw a different plugin erroring in the logs and wanted to disable it before I submitted.

paulkoan, 2018-03-23:
http://hum.is/octoprint.log

foosel, 2018-03-23:
Hm... It looks like it isn't even trying to connect to the broker. 

I can't reproduce this here though (also testing against a Mosquitto instance with plugin version 0.6). Could you share your config?

paulkoan, 2018-03-23:
Other than the hostname, everything is default, so this is all the config for the plugin:

```
  mqtt:
    broker:
      url: 10.1.1.171

```
I guess I should try and connect with a minimal paho script?

foosel, 2018-03-23:
That might certainly help narrowing that down. 

I'd have expected this output in `octoprint.log` (or alternatively some error): 

```
2018-03-23 10:27:14,746 - octoprint.plugins.mqtt - INFO - Connected to mqtt broker
```

The only reason I can think of that not being produced is either the broker URL being `None` all of a sudden (doesn't make much sense since that's checked before) or the client itself falling on its face somehow.

paulkoan, 2018-03-23:
Ok, I tried connecting direct with paho and that worked, so I resaved the config and restarted octoprint again and this time it worked!  If it fails to connect in the first instance, does it recover?  It is possible I need to restart octoprint if I change the host?

foosel, 2018-03-23:
> It is possible I need to restart octoprint if I change the host?

Yep. It's currently not reacting to settings changes directly, you'll need to restart for that. I think there was also a feature request for changing that somewhere, but right now I appear to be blind and not seeing it.

paulkoan, 2018-03-23:
Thanks for your time and plugins

closed 2018-03-23 by paulkoan