OctoPrint/OctoPrint-MQTT #36: Plugin is not showing up - AttributeError: type object 'Events' has no attribute 'CONNECTIVITY_CHANGED'

MiyazawaK, 2018-04-06:
#### What were you doing?

I've tried to install the MQTT Plugin via all 3 possible ways. It's installing properly but not showing up afterwards.

#### What did you expect to happen?

The MQTT Plugin to show up^^

#### What happened instead?

Nothing

#### Version of OctoPrint
1.3.4 (master branch)

#### Version of the MQTT plugin

current master branch (can't check via plugin manager)

#### Used MQTT broker and its version

doesnt matter here

#### Link to octoprint.log

https://pastebin.com/Fm2xnYk2

#### Link to contents of Javascript console in the browser
Starting dependency resolution...
packed_core.js:12820:9
... dependency resolution done
packed_core.js:12921:9
Initial application setup done, connecting to server...
packed_core.js:13150:9
Connected to the server
packed_core.js:11368:9
Finalizing application startup
packed_core.js:13154:17
Going to bind 29 view models...
packed_core.js:13058:13
Did not bind view model CoreWizardAclViewModel to target #wizard_plugin_corewizard_acl since it does not exist
packed_core.js:13099:29
Did not bind view model CoreWizardPrinterProfileViewModel to target #wizard_plugin_corewizard_printerprofile since it does not exist
packed_core.js:13099:29
Did not bind view model CoreWizardWebcamViewModel to target #wizard_plugin_corewizard_webcam since it does not exist
packed_core.js:13099:29
Did not bind view model CoreWizardServerCommandsViewModel to target #wizard_plugin_corewizard_servercommands since it does not exist
packed_core.js:13099:29
Did not bind view model CuraViewModel to target #wizard_plugin_cura since it does not exist
packed_core.js:13099:29
Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js:13099:29
... binding done
packed_core.js:13132:13
Application startup complete
packed_core.js:13143:13
Connected to the server

Thanks in advance :)

foosel, 2018-04-06:
❗ **[You really really really should update your OctoPrint version](https://octoprint.org/blog/2018/03/15/security-issue-update-to-1.3.6/)** ❗

And that will also make that error go away. The plugin requires OctoPrint 1.3.5 now, but I forgot to update the metadata accordingly. Plugin repository is adjusted.

MiyazawaK, 2018-04-06:
Updated to 1.3.6 and everything works fine now! Thanks a lot, especially for the quick response!

closed 2018-04-06 by MiyazawaK