OctoPrint/OctoPrint-MQTT #42: Expection in plugin when adding temprature poing

SoundGuy, 2018-05-13:
#### What were you doing?
I have both MQTT and IFTT plugins installed 

#### What did you expect to happen?

no errors. 

#### What happened instead?
errors:

the client causes something to crash with other plugins (IFTTT plugin keeps sending the OpenClient message ) and I see these errors in the log 

#### Version of OctoPrint
OctoPrint 1.3.8 running on OctoPi 0.14.0

#### Version of the MQTT plugin
MQTT (0.7.1)

#### Used MQTT broker and its version

#### Link to octoprint.log

Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/printer/standard.py", line 145, in _sendAddTemperatureCallbacks
    callback.on_printer_add_temperature(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_mqtt/__init__.py", line 204, in on_printer_add_temperature
    or abs(value["actual"] - self.lastTemp[key]["actual"]) >= threshold \
TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'
2018-05-13 15:30:53,387 - octoprint.printer.standard - ERROR - Exception while adding temperature data point to callback <mqtt.MqttPlugin object at 0xb4
03b090>
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/printer/standard.py", line 145, in _sendAddTemperatureCallbacks
    callback.on_printer_add_temperature(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_mqtt/__init__.py", line 204, in on_printer_add_temperature
    or abs(value["actual"] - self.lastTemp[key]["actual"]) >= threshold \
TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'
2018-05-13 15:30:55,370 - octoprint.printer.standard - ERROR - Exception while adding temperature data point to callback <mqtt.MqttPlugin object at 0xb4
03b090>
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/printer/standard.py", line 145, in _sendAddTemperatureCallbacks
    callback.on_printer_add_temperature(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_mqtt/__init__.py", line 204, in on_printer_add_temperature
    or abs(value["actual"] - self.lastTemp[key]["actual"]) >= threshold \
TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'
2018-05-13 15:30:56,298 - octoprint.plugins.IFTTTmaker - INFO - URL: https://maker.ifttt.com/trigger/op-ClientOpened/with/key/dUWVcmGy4B_FUwT8D0Zgit
2018-05-13 15:30:56,331 - octoprint.plugins.IFTTTmaker - INFO - Trigger: op-ClientOpened Response: Congratulations! You've fired the op-ClientOpened event Payload: {'value3': '192.168.1.108', 'value2': '', 'value1': ''}
2018-05-13 15:30:56,443 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.108
2018-05-13 15:30:56,521 - octoprint.plugins.IFTTTmaker - INFO - Event skipped: ClientClosed
2018-05-13 15:30:57,404 - octoprint.printer.standard - ERROR - Exception while adding temperature data point to callback <mqtt.MqttPlugin object at 0xb403b090>
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/printer/standard.py", line 145, in _sendAddTemperatureCallbacks
    callback.on_printer_add_temperature(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_mqtt/__init__.py", line 204, in on_printer_add_temperature

Do you need the full log or is this engouh?

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem

foosel, 2018-05-30:
I have my hands full with core OctoPrint development and haven't had a chance to look into this yet.

SoundGuy, 2018-10-11:
i'm still getting this error flooding my log. 

2018-10-11 17:42:22,490 - octoprint.printer.standard - ERROR - Exception while adding temperature data point to callback <mqtt.MqttPlugin object at 0xb307b030>
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/printer/standard.py", line 167, in _sendAddTemperatureCallbacks
    callback.on_printer_add_temperature(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_mqtt/__init__.py", line 204, in on_printer_add_temperature
    or abs(value["actual"] - self.lastTemp[key]["actual"]) >= threshold \
TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'

just curious if this is serious or not. my octoprint crashes every once in a while (week-two) and it did once in mid print and this flood of errors is making it hard to find anything else

jneilliii, 2018-10-12:
@SoundGuy, I just checked this on my OctoPrint 1.3.9 running on OctoPi 0.14.0 and am not getting this temperature error.  Not sure if this is a conflict with IFTT or maybe something was fixed between version 1.3.8 and 1.3.9 of OctoPrint that resolves this issue.

SoundGuy, 2018-10-12:
I’m on the same config. 1.3.9 with 0.14 

Oded Sharon

Email:  iPhone@odedsharon.com
Skype: soundguy27 | Twitter: @SoundGuy | Facebook:oded.sharon  | LinkedIn: SoundGuy  | Blog: http://odedsharon.com

(sent from my iPhone so probably there are some typos)

jneilliii, 2018-10-12:
Do you get the same noise in your log with the ifttt plugin disabled?

sidddy, 2018-11-04:
@jneilliii : The temperature graph handles it properly: After "host-only" startup the graph shows only the tool temperature (off). Once the MCU goes online, the bed temperature line appears.

[image]

tedder, 2018-11-04:
On it. Aside from this error, there's a really subtle error that I didn't see at first:
```
        if key not in self.lastTemp or \
            or abs(value["actual"] - self.lastTemp[key]["actual"]) >= threshold \
            or value["target"] != self.lastTemp[key]["target"]:
```

4 internet points to whoever spots it :)

tedder, 2018-11-04:
doh- actually I did that when I was doing some debugging, that's not the committed version.

jneilliii, 2018-11-05:
Double or?

tedder, 2018-11-05:
Yeah, fun times eh?

I have the fix committed, I'll make a PR after 0.8 ships.

tedder, 2018-11-05:
I'd rather roll this into 0.8.1 with other minor fixes. That way we can get the other stuff out the door first.

(can you tell I do devops/release management for a career?)

jneilliii, 2018-11-05:
No problem.  I barely understand git and have never worked with other programmers on a team, so if you see me doing something wrong on this project, or you guys want to manage it differently I'm all ears.  This one is a little different because it was left over from before, I just made minor tweaks directly to devel branch.  Moving forward I guess we should come up with a development workflow, let me know your thoughts.

sidddy, 2018-11-05:
@tedder: as far as I can see, there's a little copy-paste-error in your commit, see https://github.com/OctoPrint/OctoPrint-MQTT/commit/9fc3e60e4198f47e33038f40c4092b7b0a4fcec9#r31176383

With that propsed change, your coding is working fine for me, the exception is gone and temperatures are properly reported via mqtt.

commit 47155ec8 (2019-02-06): safer comparisons of temperature

Fixes #42, which was (float)-(None), which doesn't eval in python. Adding some safe dict navigation and also making sure we don't accidentally get caught by a floating point mismatch with a very wide threshold/sigma.
commit 2f756c35 (2019-02-06): safer comparisons of temperature

Fixes #42, which was (float)-(None), which doesn't eval in python. Adding some safe dict navigation and also making sure we don't accidentally get caught by a floating point mismatch with a very wide threshold/sigma.

patch increment of version
PR #57 (2019-02-09): fix NoneType errors on temperatures
Fixes #42 and #56, which is almost exactly the same thing. I'd forgotten this wasn't merged when @jneilliii took over the repo and released "0.8.0".

Needs some refreshed testing, please note on the PR if you have installed the PR version and it works/fails.
    
You can install this PR by using the following URL in your plugin install "from URL" panel. I think.
    
    https://github.com/OctoPrint/OctoPrint-MQTT/archive/201811_safer_temperature_evaluation_d.zip
commit f79c00bb (2019-02-09): fix NoneType temperature accesses

Fixes #42 and #56, which is almost exactly the same thing. I'd forgotten this wasn't merged when @jneilliii took over the repo and released "0.8.0".

Needs some refreshed testing, please note on the PR if you have installed the PR version and it works/fails.

You can install this PR by using the following URL in your plugin install "from URL" panel. I think.

    https://github.com/OctoPrint/OctoPrint-MQTT/archive/201811_safer_temperature_evaluation_d.zip
closed 2019-02-09 by tedder