OctoPrint/OctoPrint-MQTT #44: MQTT Broker not listening on Raspberry Pi

blandman, 2018-06-03:
I am trying to use this plugin on OctoPi from a Raspberry Pi. I've installed, rebooted, and configured the MQTT Broker to listen on the server IP and 127.0.0.1 with no luck. I am unable to connect to MQTT from any Chrome extension or Node-Red.

I ran netstat on the Pi and don't see port 1883 in use, unsure how to gather information for troubleshooting.

ukmoose, 2018-06-03:
If I understand you correctly and you cannot connect to your MQTT broker from Node-RED the problem would appear to lie with your MQTT broker and not with this OctoPrint plugin.

If you cannot connect to your broker from Node-RED the node will have a red triangle in the top right corner of the node icon.

blandman, 2018-06-03:
Hi!

Thanks for the quick response. I also tried two MQTT chrome extensions and neither worked.

I couldn’t find the port being used on the Pi in ssh, it didn’t show up in net stat

Kunsi, 2018-09-24:
Have you installed a MQTT server on the pi, or just the plugin?
The plugin itself does not include a mqtt server, instead it just connects to a existing server you have to install yourself.

blandman, 2018-09-24:
I was unaware of this, I’ll give it another go

jneilliii, 2018-10-12:
Yeah, the easiest method of getting this up if you don't have a MQTT server is to install one on the local pi.  Then the plugin will work.  Configuring an MQTT server is out of the scope of this plugin, but if you follow the instructions in Step 1 linked below. If it's not exposed to the external internet it should be safe enough, otherwise you may want to look at adding security, etc. in the config file.

https://learn.adafruit.com/diy-esp8266-home-security-with-lua-and-mqtt/configuring-mqtt-on-the-raspberry-pi#step-1-install-the-mosquitto-mqtt-components-5-3