OctoPrint/OctoPrint-MQTT #45: Query informations

Psychokiller1888, 2018-07-30:
#### What were you doing?

Developping my home assistant

#### What did you expect to happen?

#### What happened instead?

#### Version of OctoPrint

1.3.9

#### Version of the MQTT plugin

0.7.1

#### Used MQTT broker and its version

#### Link to octoprint.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem

How do we query data? It's good to have randomly heartbeat, but querying the progress and others would make sense?

foosel, 2018-08-01:
Read the README again, it contains information on which topics "progress and others" are published. MQTT is a publish/subscribe protocol, there's no querying of data, there's only subscribing to topics and waiting for events to be published on them.

closed 2018-08-01 by foosel

Psychokiller1888, 2018-08-01:
No....

First of all, the readme doesn't give al lthe topics, found them though

MQTT can be bidirectional, meaning your plugin could listen for messages sent to it from another devices and send out another message...

foosel, 2018-08-01:
The plugin doesn't subscribe to any topics, it only publishes (otherwise that would be documented). There already is an existing feature request for changing that, waiting for someone to implement it.

If there are any topics missing from the documentation please send a PR to fix this.

Psychokiller1888, 2018-08-01:
Waiting for someone to implement it... Ok.... Well, I'll do it myself then, np

foosel, 2018-08-01:
> Well, I'll do it myself then, np 

That's the spirit! 👍