OctoPrint/OctoPrint-MQTT #46: Flooding the broker with "octopi-ender3/temperature/tool0"

StefanIGit, 2018-09-23:
#### What were you doing?
Hmm I think ;)
- installed the mqtt plugin
- configured it 
- printed
- played with node red while printing
- print finished
- switch the printer  off (not the PI)
- next day (now) played more with mqtt and node red
- changed the octopi hostname and Rebooted

At some point I did a 
   mosquitto_sub -h IP -t '#'
and the terminal get flooded with this
   octopi-ender3/temperature/tool0 {"_timestamp":1537650653,"actual":22.15,"target":0}

Since my printer is still powered off I assume the is the last state from the day before, before I powered it off.

#### What did you expect to happen?

no message since the printer is disconnected

#### What happened instead?

but definitely not a all those messages and surely not in a flooding/spamming fashion

#### Version of OctoPrint

OctoPrint 1.3.9 auf OctoPi 0.15.0

#### Version of the MQTT plugin
0.7.1
#### Used MQTT broker and its version
pi@octopi-ender3:~ $ apt show mosquitto
Package: mosquitto
Version: 1.5-0mosquitto2

#### Link to octoprint.log

https://gist.github.com/StefanIGit/f8c2064d6057dc7b51a1a4ae3c45f9e3

foosel, 2018-09-24:
How do you define "flooding/spamming fashion"? The data should actually only be sent if there's been a change to it, so clearly something's up here since when the printer is disconnected there also shouldn't be a change of course. Still, would be interesting to know how fast the messages were arriving. Was the `_timestamp` field changing or did it stay static?

StefanIGit, 2018-09-24:
Hi, it was the same line with static timestamp. 
I started the command "mosquitto_sub -h IP -t '#'"  and I got pages ang pages of the same line until I crtl+c it.
I tried to reproduce it but was not able yet, maybe I need a longer print for that.

mkh595, 2018-10-18:
> octopi-ender3/temperature/tool0
 {"_timestamp":1537650653,"actual":22.15,"target":0}
Since my printer is still powered off I assume the is the last state from the day before, before I powered it 
off.

[image]

Message was sent with RETAIN flag. Subscriber will receive it each time when subscribe to this topic. It can be deleted from MQTT broker only manually.

tedder, 2019-09-23:
Not reproducable by original reporter; if this is a `retain` issue, it is configurable by #55 now in v0.8.4.

closed 2019-09-23 by tedder