OctoPrint/OctoPrint-MQTT #47: Up for adoption

foosel, 2018-09-24:
I have my hands way too full with OctoPrint's maintenance to give this plugin the attention it needs. Hence I'm looking for a new maintainer to adopt it.

tedder, 2018-09-24:
I would like to take this, or share the ownership if anyone else is interested.

foosel, 2018-09-24:
@Kunsi already raised her hand, but I'm not sure, maybe she'd also be interested in shared ownership. I'll leave that for y'all to figure out ;) 

In any case, thanks for the interest! :)

Kunsi, 2018-09-24:
Any help is welcome, so feel free to join the team :)

foosel, 2018-09-24:
@Kunsi you should have admin permissions, so you should be able to invite @tedder as a collaborator

Kunsi, 2018-09-24:
Oh, yeah. Done that now :3

jneilliii, 2018-09-25:
I'd like to be part of this team too if that's cool with you guys.

tedder, 2018-09-25:
jneilliii is pretty much the octoprint plugin god.

jneilliii, 2018-09-25:
You're funny Ted, I just hack things together.

jneilliii, 2018-09-26:
Thanks for the invite, let me know how I can help.  I'll probably start by going through and flushing out some of the issues. Adoption complete, readme updated removing adoption notice.

closed 2018-09-26 by jneilliii