OctoPrint/OctoPrint-MQTT #48: configurable _timestamp field name?

tedder, 2018-09-27:
I'm pushing output of this plugin into Elasticsearch, which doesn't like having a field use its internal name of _timestamp.

Perhaps the name of that field could be configurable? I might do it but I have a bunch of loose ends, so I wanted to at least place it here.

tedder, 2018-09-27:
wondering if @foosel has seen this since they use ELK.

RedWagon, 2019-02-02:
+1 to this, I'm having issues using the timestamp in AWS IoT due to the SQL selection language used to interpret MQTT payloads when executing rules.

jneilliii, 2019-04-19:
Incorporated into branch 1.8.2 for testing.  Install using the following url in plugin manager to verify.

https://github.com/OctoPrint/OctoPrint-MQTT/archive/0.8.2.zip

PR #60 (2019-05-19): 0.8.2
- Add option to statically set client_id for MQTT client connection to broker, resolves #59 
- Make timestamp field name configurable, resolves #48
closed 2019-05-19 by jneilliii