OctoPrint/OctoPrint-MQTT #53: Progress Topic not published since last update

PeggyFree, 2018-11-14:
Hello, first of all thank you for this plugin.

I've been using intensively this MQTT plugin, retrieving most data in my automation system. Since last update I don't get octoprint/progress messages anymore. 
I get many polluting "positionupdate" and "zchange" messages I would like to get rid of, but progress messages are gone. What happened?

tedder, 2018-11-14:
@PeggyFree what happened? I assume you resolved it in some way.

PeggyFree, 2018-11-15:
@tedder Thank you for your interest. I discovered that it is the new additional data in the 'progress' topic that makes my python client crash when decoding the message. I disabled this option until I'll have time to look at.