OctoPrint/OctoPrint-MQTT #55: Feature request: Configure retain flag (for AWS IoT)

RedWagon, 2019-02-02:
#### What were you doing?

Publishing to AWS IoT via mosquitto broker 
https://aws.amazon.com/blogs/iot/how-to-bridge-mosquitto-mqtt-broker-to-aws-iot/

#### What did you expect to happen?

AWS IoT would receive the messages

#### What happened instead?

All messages from the mosquitto broker stopped being processed by AWS, even ones unrelated to OctoPrint.

#### Version of OctoPrint

1.3.10

#### Version of the MQTT plugin

0.8.0

#### Used MQTT broker and its version

mosquitto 1.5.5

#### Workaround:
Clone plugin source 
```
cd OctoPrint-MQTT/octoprint_mqtt/
sed -i 's/retained=True/retained=False/' __init__.py
sed -i 's/retain=True/retain=False/' __init__.py
```
Install plugin

jneilliii, 2019-04-16:
@RedWagon, would making this an optional setting work for you?  I'd be concerned with some of the messages being retained due to the discussion over in #46.

tedder, 2019-05-20:
[AWS docs support the need to not retain *anything*](https://docs.aws.amazon.com/iot/latest/developerguide/mqtt.html):
> The MQTT specification provides a provision for the publisher to request that the broker retain the last message sent to a topic and send it to all future topic subscribers. AWS IoT does not support retained messages. If a request is made to retain messages, the connection is disconnected.

It'll definitely need to be a setting.

PR #63 (2019-09-23): push 'retain' flag to publish-time only
- instead of passing in a retain flag, move it to a config option and don't even determine what to do until publish-time. If we want it to be more granular later, we can, but it needs more info than "boolean" to capture it- e.g. we need to capture "do whatever the default is" in addition to a bool. yolo.
- add to template
- add retain debug line
- make sure to retain on the queue too

@jneilliii please review; the log entry shows 'true' when expected, but `mosquitto_sub`'s %r option never changes from nonzero. I need to look at it more, unless you see so [... cut]
closed 2019-09-23 by tedder