OctoPrint/OctoPrint-MQTT #56: Error in temp-calculation halts MQTT sending

joshiegy, 2019-02-06:
#### What were you doing?

The plugin cant seem to send data over MQTT since a variable cant be calculated

1. install plugin
2. setup against working MQTT
3. Start a print

#### What did you expect to happen?
Not this error in the log

#### What happened instead?
octoprint     | 2019-02-06 18:22:35,557 - octoprint.printer.standard - ERROR - Exception while adding temperature data point to callback <mqtt.MqttPlugin object at 0x7f083933d310>                               
octoprint     | Traceback (most recent call last):
octoprint     |   File "/opt/octoprint/venv/lib/python2.7/site-packages/octoprint/printer/standard.py", line 167, in _sendAddTemperatureCallbacks                                                                 
octoprint     |     callback.on_printer_add_temperature(data)
octoprint     |   File "/opt/octoprint/venv/lib/python2.7/site-packages/octoprint_mqtt/__init__.py", line 213, in on_printer_add_temperature                                                                      
octoprint     |     or abs(value["actual"] - self.lastTemp[key]["actual"]) >= threshold \                                                                                                                         
octoprint     | TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'

#### Version of OctoPrint
1.3.10

#### Version of the MQTT plugin
0.8.0

#### Used MQTT broker and its version
Mosquitto, unkown (latest docker container toke/mosquitto around 2018-12-01 (

#### Link to octoprint.log
These lines repeat once per second and nothing else
octoprint     | 2019-02-06 18:22:35,557 - octoprint.printer.standard - ERROR - Exception while adding temperature data point to callback <mqtt.MqttPlugin object at 0x7f083933d310>                               
octoprint     | Traceback (most recent call last):
octoprint     |   File "/opt/octoprint/venv/lib/python2.7/site-packages/octoprint/printer/standard.py", line 167, in _sendAddTemperatureCallbacks                                                                 
octoprint     |     callback.on_printer_add_temperature(data)
octoprint     |   File "/opt/octoprint/venv/lib/python2.7/site-packages/octoprint_mqtt/__init__.py", line 213, in on_printer_add_temperature                                                                      
octoprint     |     or abs(value["actual"] - self.lastTemp[key]["actual"]) >= threshold \                                                                                                                         
octoprint     | TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem
N/A

jneilliii, 2019-02-06:
@tedder, isn't this something you had a fix for?

tedder, 2019-02-06:
Yeah, I thought we'd folded it in, but the branch still exists: 386b0d5587adda908c5c4b4552f5085689f0ea37

Certainly I don't see the changes based on that exception. I'll get it up to date and PR it.

tedder, 2019-02-09:
fixed+released as 0.8.1.

closed 2019-02-09 by tedder