OctoPrint/OctoPrint-MQTT #58: Only Firmware Error Events - Question

ZhuDaHai, 2019-04-06:
Is it possible to only publish a Firmware Error event?
I am just wanting to notify the MQTT broker if/when there is a firmware error (printer halt, thermal runaway, etc). 
I see there is a Printer Data option, but I don't need all that constant traffic. I just want to know if there was an error.

Thanks!

tedder, 2019-04-06:
Have you muted all of the topics you aren't interested in already?

ZhuDaHai, 2019-04-07:
Yes, I believe so, but the documentation v. the plugin screens are confusing.
I have only 'Unclassified Events' checked under 'Activate Event Messages', but I can't find any information on what that includes.
The progress Topic contains the 'Error' event that I want, but it also includes dozens of other events I don't want/need. 
It's difficult to cause a Print Error event, so I'd like to know if its even possible to do what I want first, before having to throw darts at it and then find it out can't be done.

ZhuDaHai, 2019-04-07:
I tested the Topics and events with Temperature Messages activated and saw them come into the MQTT broker using MQTT Explorer. So I know the connection was good and messages were being sent.
Then with 'Activate Event Messages' checked and only 'Unclassified Events' checked, i sent a M112 to the Smoothieware firmware through the OctoPrint Terminal. OctoPrint went offline with "Offline (Error: Closing serial port due to emergency stop M112)", but no message was sent via MQTT. This was the type of Error I was hoping to catch in the MQTT broker so I could run an automation off it, but it was never sent.
So, how do I set this OctoPrint Plugin to send only Error Events over MQTT?
Thanks.

ZhuDaHai, 2019-04-07:
Ok, looked through the code and found what I wanted:
```
Communication Event:
   Error = {"_timestamp": 1554627685, "reason": "firmware", "_event": "Error", "error": ""}
```
That narrows it down enough that I can use it.

closed 2019-04-07 by ZhuDaHai