OctoPrint/OctoPrint-MQTT #61: Undesired messages for non-existing chamber temperature

gandy92, 2019-05-15:
#### What were you doing?

* Updated to latest OctoPrint release
* Installed and configured the MQTT plugin
* Watched the topics with `mosquito_sub -v -t octoprint/#`

The printer profile correctly reflects the missing heated chamber:
[image]

#### What did you expect to happen?
temperate messages for `bed` and `tool0` only

#### What happened instead?
regular temperate messages for `chamber` with updating `_timestamp` readings and temperature values `null`

#### Version of OctoPrint

OctoPrint 1.3.11 auf OctoPi 0.16.0

#### Version of the MQTT plugin

0.8.1

#### Screenshot showing the problem

```
octoprint/temperature/chamber {"_timestamp": 1557933342, "actual": null, "target": null}
octoprint/temperature/bed {"_timestamp": 1557933342, "actual": 24.22, "target": 0.0}
octoprint/temperature/chamber {"_timestamp": 1557933345, "actual": null, "target": null}
octoprint/temperature/bed {"_timestamp": 1557933345, "actual": 23.98, "target": 0.0}
octoprint/temperature/chamber {"_timestamp": 1557933348, "actual": null, "target": null}
octoprint/temperature/tool0 {"_timestamp": 1557933348, "actual": 23.24, "target": 0.0}
octoprint/temperature/chamber {"_timestamp": 1557933351, "actual": null, "target": null}
octoprint/temperature/tool0 {"_timestamp": 1557933351, "actual": 23.52, "target": 0.0}
octoprint/temperature/bed {"_timestamp": 1557933351, "actual": 24.14, "target": 0.0}
octoprint/temperature/chamber {"_timestamp": 1557933354, "actual": null, "target": null}
octoprint/temperature/bed {"_timestamp": 1557933354, "actual": 23.91, "target": 0.0}
octoprint/temperature/chamber {"_timestamp": 1557933357, "actual": null, "target": null}
octoprint/temperature/bed {"_timestamp": 1557933357, "actual": 24.1, "target": 0.0}
```

foosel, 2019-05-16:
`chamber` should behave the same as `bed` in all regards.

jneilliii, 2019-05-19:
@tedder, not sure that would work since chamber is a sub-key of temperature and therefore would fall under the temperatureActive setting. I think to handle this it would be best to look inside the temperature publishing part [here](https://github.com/OctoPrint/OctoPrint-MQTT/blob/5957789c71142c81091d0a3b86ef2867a897b7ec/octoprint_mqtt/__init__.py#L215). If we do I assume we should also expand for bed since there may not be a heated bed either on the printer.  Is that even picked up now from the existing code or not? I've always had a heated bed, so wasn't sure.

jneilliii, 2019-05-19:
Now that I've fully looked at the code you linked to in regards to masking the temperature data, there is no reason that I can tell OctoPrint would be sending this chamber data to the MQTT plugin at all unless there's a flaw in the [on_printer_add_temperature](https://github.com/OctoPrint/OctoPrint-MQTT/blob/5957789c71142c81091d0a3b86ef2867a897b7ec/octoprint_mqtt/__init__.py#L210) callback of the mixin.

tedder, 2019-05-20:
I'm just starting to upgrade my printers to '11 and see what happens. I'm not seeing it yet. I'm going to fiddle with settings a little bit more, though.

tedder, 2019-05-20:
I'm seeing it, I just had my "scrolling mqtt status" screen muting too many fields. Here it is:

```
octoprint/prusamk3/temperature/bed {"_timestamp": 1558320950, "actual": 42.9, "target": 0.0}
octoprint/prusamk3/temperature/tool0 {"_timestamp": 1558320952, "actual": 53.9, "target": 0.0}
octoprint/prusamk3/temperature/chamber {"_timestamp": 1558320953, "actual": null, "target": null}
```

I know that I have "chamber" unchecked in my print profile.

tedder, 2019-05-20:
Probably gonna be easiest to change the iterator in on_printer_add_temperature. Instead of looking for n-1 keys (since 'time' is one) and then perhaps subtracting bed and chamber, figure we can just unroll the loop and handle each of the up to 3 entries individually, especially because we can grab the configs:

```
    self._logger.info("HB? {}".format( self._printer_profile_manager.get_current_or_default()["heatedBed"] ))
    self._logger.info("HC? {}".format( self._printer_profile_manager.get_current_or_default()["heatedChamber"] ))
```

idcrook, 2019-05-25:
I was also noticing these spurious `chamber` temperature updates, and found this issue.

```
octoprint/rpip10/temperature/chamber {"_timestamp": 1558815704, "actual": null, "target": null}
```

thanks for getting the fix in

PeggyFree, 2019-09-23:
Thank you by advance for publishing this fix.

jneilliii, 2019-09-23:
@tedder, I don't think you ever merged this pull request....

tedder, 2019-09-23:
4 months ago:
> I'm going to let this bake on my Pis for a day or two.

^-^

commit 9a3ca7ea (2019-05-20): mask empty temperatures, eg bed or chamber

- Fixes #61 and probably fixes a case where the bed isn't heated too.
- use `six` to check for stringiness. Should help with py2/py3.
- adjusted some whitespace per `flake`.
- prep for release
PR #62 (2019-09-23): mask empty temperatures, eg bed or chamber
- Fixes #61 and probably fixes a case where the bed isn't heated too.
- use `six` to check for stringiness. Should help with py2/py3.
- adjusted some whitespace per `flake`.
- prep for release

I'm going to let this bake on my Pis for a day or two.
commit e8696675 (2019-09-23): mask empty temperatures, eg bed or chamber

- Fixes #61 and probably fixes a case where the bed isn't heated too.
- use `six` to check for stringiness. Should help with py2/py3.
- adjusted some whitespace per `flake`.
- prep for release
closed 2019-09-23 by tedder