OctoPrint/OctoPrint-MQTT #64: Install plugin

GrRivero, 2019-08-05:
#### What were you doing?

Install OctoPrint-MQTT via pluging manaer and command line.

#### What did you expect to happen?

Install plugin.

#### What happened instead?

Error log:

pi@raspberrypi:~ $ pip install "https://github.com/OctoPrint/OctoPrint-MQTT/archive/master.zip"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/OctoPrint/OctoPrint-MQTT/archive/master.zip
  Downloading https://github.com/OctoPrint/OctoPrint-MQTT/archive/master.zip
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/_internal/cli/base_command.py", line 143, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/_internal/commands/install.py", line 338, in run
    resolver.resolve(requirement_set)
  File "/usr/lib/python2.7/dist-packages/pip/_internal/resolve.py", line 102, in resolve
    self._resolve_one(requirement_set, req)
  File "/usr/lib/python2.7/dist-packages/pip/_internal/resolve.py", line 256, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/usr/lib/python2.7/dist-packages/pip/_internal/resolve.py", line 209, in _get_abstract_dist_for
    self.require_hashes
  File "/usr/lib/python2.7/dist-packages/pip/_internal/operations/prepare.py", line 283, in prepare_linked_requirement
    progress_bar=self.progress_bar
  File "/usr/lib/python2.7/dist-packages/pip/_internal/download.py", line 836, in unpack_url
    progress_bar=progress_bar
  File "/usr/lib/python2.7/dist-packages/pip/_internal/download.py", line 673, in unpack_http_url
    progress_bar)
  File "/usr/lib/python2.7/dist-packages/pip/_internal/download.py", line 897, in _download_http_url
    _download_url(resp, link, content_file, hashes, progress_bar)
  File "/usr/lib/python2.7/dist-packages/pip/_internal/download.py", line 619, in _download_url
    consume(downloaded_chunks)
  File "/usr/lib/python2.7/dist-packages/pip/_internal/utils/misc.py", line 866, in consume
    deque(iterator, maxlen=0)
  File "/usr/lib/python2.7/dist-packages/pip/_internal/download.py", line 585, in written_chunks
    for chunk in chunks:
  File "/usr/lib/python2.7/dist-packages/pip/_internal/download.py", line 574, in resp_read
    decode_content=False):
  File "/usr/share/python-wheels/urllib3-1.24.1-py2.py3-none-any.whl/urllib3/response.py", line 494, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/usr/share/python-wheels/urllib3-1.24.1-py2.py3-none-any.whl/urllib3/response.py", line 459, in read
    raise IncompleteRead(self._fp_bytes_read, self.length_remaining)
  File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
    self.gen.throw(type, value, traceback)
  File "/usr/share/python-wheels/urllib3-1.24.1-py2.py3-none-any.whl/urllib3/response.py", line 374, in _error_catcher
    raise ReadTimeoutError(self._pool, None, 'Read timed out.')
ReadTimeoutError: HTTPSConnectionPool(host='codeload.github.com', port=443): Read timed out.

#### Version of OctoPrint

1.3.11

#### Version of the MQTT plugin

None

tedder, 2019-08-05:
Can you fix the issue title?

In any case, if you note the error at the end, it's either transient or you have a local issue. This should not time out:
```
curl -D - https://codeload.github.com/
```

GrRivero, 2019-08-06:
I'm sorry, false issue.

Other pluggins are installed ok but the only has problem its this in my octoprint.

I'm fixed the issue restarting the router.

Sorry and thanxs.

closed 2019-08-06 by GrRivero