OctoPrint/OctoPrint-MQTT #66: Missing the option for "clean session"

robalcnj, 2019-09-26:
Some brokers like mqtt.thingspeak.com require a flag "clean session=true" to be sent at authentication or it will return a permanent not authorised.

jneilliii, 2019-09-28:
Can you provide a link to the docs on thingspeak's site relative to this flag?

jneilliii, 2019-09-28:
I've added a pull request that adds this option.  I'm able to successfully connect now to thingspeak in my testing, but setting up the topics is going to be a challenge I think as it requires the API key to be added to the end of every topic your publishing to.  I assume you have more experience with Thingspeak since I just created the account to test the connection. You can install using the url below to test out yourself while the PR is under review.

https://github.com/OctoPrint/OctoPrint-MQTT/archive/0.8.5.zip

PR #67 (2019-09-28): 0.8.5
Added clean_session option for Broker, resolves #66.
closed 2019-09-28 by jneilliii