OctoPrint/OctoPrint-MQTT #68: 0.8.5 breaks existing client_id configuration

oxivanisher, 2019-09-29:
#### What were you doing?

Updated the plugin to 0.8.5

#### What did you expect to happen?

The plugin keeps working

#### What happened instead?

The plugin stopped working due to the last changes.

#### Version of OctoPrint

OctoPrint 1.3.11 running on OctoPi 0.15.1

#### Version of the MQTT plugin

0.8.5

#### Used MQTT broker and its version

Docker container from dockerhub: cyrilix/rabbitmq-mqtt

#### Link to octoprint.log

```
2019-09-29 10:50:23,839 - octoprint.plugin - ERROR - Error while calling plugin mqtt
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 219, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_mqtt/__init__.py", line 82, in on_startup
    self.mqtt_connect()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_mqtt/__init__.py", line 299, in mqtt_connect
    self._mqtt = mqtt.Client(client_id=client_id, protocol=protocol, clean_session=clean_session)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/paho/mqtt/client.py", line 511, in __init__
    raise ValueError('A client id must be provided if clean session is False.')
ValueError: A client id must be provided if clean session is False.
```

#### More description
If i remember correctly, the client id was configured before the update. It seems, the update process deleted this setting (problem A) and then does not tell the user, that the settings are not configured as they should (problem B).

jneilliii, 2019-09-29:
Thanks for the post and detailed errors. Relative to this issue is that error specific to your mqtt broker or part of the mqtt standard? I did not get those errors while testing with mosquito.

jneilliii, 2019-09-29:
I've verified the update made no change whatsoever to the client id, so I'm not sure what happened there for you.  I just released version 0.8.6 that swaps the default setting for clean_session to true which should prevent this error for other users and should be the default setting based on paho client commands anyway. 

It would make sense to do some UI enhancements to tie client_id and clean_session options together in some way and enable the client_id option by default when clean_session is unchecked, but wanted to get the update out as quickly as possible.

oxivanisher, 2019-09-29:
Thank you very much. If I can help you test something, just tell me.

(My broker is a container instance of cyrilix/rabbitmq-mqtt with nothing special configured.)

closed 2019-09-29 by oxivanisher