OctoPrint/OctoPrint-MQTT #70: Not possible to get Startup event?

Lenbok, 2019-10-09:
#### What were you doing?

I am trying to create a node-red flow to execute an action when Octoprint starts up and shuts down. I happily receive Shutdown events over MQTT, but it appears that no Startup event is sent by the MQTT plugin. 

1. Configure the MQTT plugin to send messages to your broker
2. Subscribe to octoprint/+/event/+ messages using your favourite tool.
3. Start Octoprint.

#### What did you expect to happen?

We should see a Startup message.

#### What happened instead?

No Startup message. Other events such as ClientOpened/Shutdown etc arrive when expected.

#### Version of OctoPrint

1.13.11

#### Version of the MQTT plugin

0.8.6

#### Used MQTT broker and its version

mosquitto 1.4.14

#### Link to octoprint.log

[octoprint.log](https://github.com/OctoPrint/OctoPrint-MQTT/files/3706411/octoprint.log)

#### Link to contents of Javascript console in the browser

No web browser involved.

#### Screenshot(s)/video(s) showing the problem

colinl, 2020-01-11:
You can use the LWT connected notification to know when Octoprint connects to MQTT.

Lenbok, 2020-01-11:
Yes, that's what I've done as a workaround in my Node-RED flow.  It's ugly and not the expected behaviour, so hopefully there is a proper fix.

smurfix, 2020-05-14:
@Lenbok This may be a stupid question, but what benefit would you get from an explicit Startup event that the LWT topic doesn't already give you?

Lenbok, 2020-05-15:
Octoprint MQTT events contain timestamps (using the time on the octoprint host), which I've been using, but LWT topics don't have timestamps. If the time gets out of sync between the printer host and my Node-RED server, the flows won't work as expected.

smurfix, 2020-05-15:
A server's LWT topic is up-to-date by definition, thus why would it need a timestamp?
It'd be trivial to JSONify the message content and add one, of course …

Lenbok, 2020-05-15:
Don't put words in my mouth - I wasn't implying that a LWT message should have a timestamp. I was explaining that the explicit Startup event *would* have a timestamp, and it would be according to the same clock as the existing octoprint events (whereas retrofitting a timestamp would be according to the clock wherever you're doing the retrofitting).