OctoPrint/OctoPrint-MQTT #71: How do I debug?

ir-fuel, 2019-10-11:
I have everything correctly configured (AFAIK), I have other systems writing to my MQTT server (running on another machine, not locally on the octoprint Raspberry Pi) so I know that part is working, yet I get zero info in there from the plugin.

I tried doing a `tail` on the octoprint log file (whilst setting the debug level of the plugin to INFO) but I see nothing appearing there.

Any suggestions on how to track what might be happening?

Thanks

jneilliii, 2019-10-12:
Assuming you've configured your broker correctly you can use an mqtt client software to subscribe to the configured base topic and see the messages it is sending. For example, If the base topic is octoprint then subscribing to octoprint/# will show all messages, at least that's how mqtt spy works.

ir-fuel, 2019-10-12:
Yes I know. The issue is nothing is appearing. So I was wondering how I can figure out what's wrong, since the logs don't say anything either.

tedder, 2019-10-16:
Share your `octoprint.log`. All of it. Otherwise we're just guessing.

ir-fuel, 2019-10-17:
Here you go
[octoprint.log](https://github.com/OctoPrint/OctoPrint-MQTT/files/3738723/octoprint.log)

tedder, 2019-10-17:
```
2019-10-17 10:28:21,801 - octoprint.plugins.mqtt - INFO - Connected to mqtt broker
2019-10-17 10:30:21,346 - octoprint.plugins.mqtt - ERROR - Disconnected from mqtt broker for unknown reasons (network error?), rc = 1
2019-10-17 10:30:21,379 - octoprint.plugins.mqtt - ERROR - Disconnected from mqtt broker for unknown reasons (network error?), rc = 1
2019-10-17 10:30:24,439 - octoprint.plugins.mqtt - INFO - Connected to mqtt broker
```

Also what's your log level (in config->logging, bottom right of that page)?

ir-fuel, 2019-10-18:
No idea what changed, but all of the sudden the print progress started appearing in my node-red dashboard. Seems the mqtt connection works now.

closed 2019-10-18 by ir-fuel