OctoPrint/OctoPrint-MQTT #74: Message publishing too fast

ut1545, 2019-11-19:
What were you doing?
I'm using published state change messages into openHab in order to take some actions based on their content but, in some cases, the time between messages are too short. For example, between a FINISHED and OPERATIONAL state messages the time passed is about 90 milliseconds so, openHab can't catch all of them. 

What did you expect to happen?
I'd like to suggest that a minimum time between messages be introduced as a parameter option. A delay of 500 milliseconds could be nice.

What happened instead?

Version of OctoPrint: 1.3.10

Version of the MQTT plugin: 0.8.1

Kunsi, 2019-11-19:
I'd suggest (also) reporting this to OpenHAB, since Octoprint-MQTT isn't the only tool I know publishing *a lot* of messages per second. OpenHAB should be able to catch all those messages.

ut1545, 2019-11-19:
Thanks. I will also do that.