OctoPrint/OctoPrint-MQTT #75: Shutdown (or similar events generated on shutting down octopi) not working reliably

spamik, 2020-01-07:
I'm trying to work in my automation system with shutdown event send from octoprint, but it's not working very properly. Shutdown event is sent to MQTT in about 50% cases. Well, to be honest, I have 2 octopi servers, one running on old Raspberry 2B, second on Raspberry 3B. On Raspberry 2B it seems is working OK, on 3B sometimes yes, sometimes no.

I'm not sure what cause this. My idea is that 3B is faster and it's shutting down faster. And maybe it's possible that in this case octoprint server is exited before MQTT plugin catches shutdown event and send it via MQTT? If it's true, would be possible in shutdown procedure of MQTT plugin just wait little time to process generated events before disconnecting from broker and quitting?

colinl, 2020-01-11:
You can use the LWT disconnected notification to know when the server disconnects from MQTT.

spamik, 2020-01-11:
Yeah, I thinked about that but it's not entirely the same. In normal condition it will be working fine. But I have little worries that if something happen like WiFi issue, MQTT broker restart or something like that, this event will be triggered even if printer is running (and it can cut off power to running printer :-) ).

colinl, 2020-01-11:
Good point, I don't know enough about how the code works to know how to fix that (presumed) race condition.

douvers, 2022-05-28:
I also have found that the mqtt event Shutdown is unreliable as it is not always published. Hopefully this issue will get some attention someday.