OctoPrint/OctoPrint-MQTT #76: Retain messages won't make sense for events

spamik, 2020-01-07:
I believe that events messages should be send without retain flag even if it's configured in settings. Now all of them are sent with retain and this setting is turned on by default.

Maybe it will have some sense when events will have some common topic but when each event has it's own topic, newly connected broker receive a lot of messages with opposite meaning (like Client Opened and Client Disconnected, Shutdown event when octoprint is running and so on).

Kllrv, 2020-01-19:
I agree, and furthermore all the different topics being sent at once to new clients makes no sense. It's very hard to make anything react to different events in a meaningful manner like this.

smurfix, 2020-05-14:
Well, that's a question of event design. Right now we have events (like PrintDone or UserLoggedIn) which make no sense to retain (the fact that I last paused a print two months ago is not interesting), but others (temperature, Z Layer, FirmwareData) are more like a state and thus should have Retain set.

My idea would be to move the State-like events to a separate subtopic. Also, we should use "clean" stateful topics to reflect some events. I.e. instead of PrintStarted and PrintStopped events (you need to compare timestamps to extract meaning from them when you start up your client, ugh!) there should be a PrinterState that contains state=printing file=foobar.gcode, or state=finished, state=halted, state=paused.

Side benefit, that way the printer state can be reflected in Home Assistant (or whatever) directly, instead of depending on custom code that does event filtering and timestamp comparison, and which will break if somebody decides to add a new state.

PR #100 (2024-01-07): Improvement to retain usage
A few general improvements to the way retain is used, fixing a couple of bugs.

- LWT topic now has it's own setting for retain. Generally even when you chose NOT to retain other topics, the LWT topic is desired to always be retained since any clients listening to the topic will need to know if the device has an active session with the MQTT server.

CLOSES: #94, #72

- Fixing support for use of the **retained** argument in calls to `mqtt_publish()`. To preserve current behaviour to any callers which do not specify this param its default value is None which will reve [... cut]
closed 2024-01-07 by jneilliii