OctoPrint/OctoPrint-MQTT #77: LWT Connected notification not working following temporary network failure.

colinl, 2020-01-11:
#### What were you doing?

A temporary failure of the network connection to the (remote) MQTT server occurred.

Ideally provide exact steps to follow in order to reproduce your problem: -->

1. Connect Octoprint to a remote MQTT server and enable LWT.
2. Connect an mqtt client to the MQTT server and subscribe to the LWT topic, the 'connected' message is seen (provided Retain is selected in Octoprint).
3. Disconnect the network between Octoprint and the MQTT server, whilst not affecting the connection  between the client and the server.
4. Wait for the 'disconnected' message in the client (this can take some time depending on the Keep Alive setting in Octoprint).
5. Restore the network connection.

#### What did you expect to happen?
The MQTT client should see a 'connected' message

#### What happened instead?
Octoprint connects ok but does not publish the 'connected' message.

#### Version of OctoPrint
1.3.12

#### Version of the MQTT plugin

0.8.6

#### Used MQTT broker and its version
Mosquitto 1.5.7

#### Link to octoprint.log

https://gist.github.com/colinl/da82221f34f060dcf5d4b8404ad9967d

#### Comments
I believe that the problem is that the LWT publish is executed from within mqtt_connect(), but this is not called following a re-connection.  I think instead that the publish should be performed in _on_mqtt_connect().

I am in the process of testing this and expect to submit a PR shortly.

colinl, 2020-01-11:
When I go to generate a PR it says I should do this against the devel branch, but when I look at the history that branch does not seem to be up to date.  Should I generate it against master or devel?

jneilliii, 2020-01-11:
Generate against master, that template was left over from when foosel maintained the repo.

colinl, 2020-01-11:
OK, will do.  Thanks.

smurfix, 2020-05-14:
Shouldn't this issue be closed?