OctoPrint/OctoPrint-MQTT #79: Support for OctoPrint 1.4 Python 3

datag, 2020-01-19:
On my experimental OctoPrint 1.4.0-rc3 installation using Python 3 the log outputs

> octoprint.plugin.core - WARNING - Plugin MQTT (0.8.6) is not compatible to Python 3.7.3 (compatibility string: >=2.7,<3).

Are you planning compatibility w/ Python 3? If yes, please let me know whether you need help implementing or testing it.

tedder, 2020-01-19:
we need to! feel free if you want to dig in.

jneilliii, 2020-01-19:
I think it works out of the box, but I'll need to double check. You can force compatibility by setting a flag in your config.yaml to get it to load. 

```
plugins:
  _forcedCompatible:
    - mqtt
```

datag, 2020-01-19:
@jneilliii Thanks for that hint. With this setting I can confirm that the plugin loads and its basic functionality is working.

@tedder So the plugin needs to be tested in detail and "just" needs to be declared as compatible with Python 3? In this case it must be ensured that no incompatible changes (both Python 2 and Python 3) are introduced as long as Python2-support isn't dropped?

tedder, 2020-01-19:
Once it's declared it may be compatible. Seems like there's always a byte vs string type of issue we forget. So reporting to us or fixing if you find anything is great. I probably won't be able to run 1.4.0 until it is out of RC.

blalor, 2020-05-09:
I think this is fixed by #84?

jneilliii, 2020-05-09:
I started that PR based on my 0.8.7 branch in preparation to merge in to master once we've had more testers. Closed #83 because it was based on an old dev version. If you want to test it out and let us know that would be great. I was about to do some testing but I don't really use it typically so don't have integrations/etc like others may have.

datag, 2020-06-30:
I can confirm that w/ 0.8.7 the plugin is loaded successfully and basic functionality is working w/ OctoPrint 1.4.0 (stable) without the need of `_forcedCompatible`. Thanks!

closed 2020-06-30 by jneilliii