OctoPrint/OctoPrint-MQTT #82: [Feature Request] only publish temperatures with significant changes

CWempe, 2020-02-16:
It would be nice if you could define how often the temperatures should be published.

Like "only if full degree changes" (55° -> 56°) or every 10 seconds.

Currently my smart home log (openHAB) updates multiple times per minute even if the changes are insignificant.

```
2020-02-16 18:56:04.300 [vent.ItemStateChangedEvent] - OctoprintPrinterBedTemp changed from 55.22 °C to 55.34 °C
2020-02-16 18:56:12.286 [vent.ItemStateChangedEvent] - OctoprintPrinterBedTemp changed from 55.34 °C to 55.21 °C
2020-02-16 18:56:14.301 [vent.ItemStateChangedEvent] - OctoprintPrinterBedTemp changed from 55.21 °C to 55.08 °C
2020-02-16 18:56:18.296 [vent.ItemStateChangedEvent] - OctoprintPrinterBedTemp changed from 55.08 °C to 54.96 °C
2020-02-16 18:56:20.315 [vent.ItemStateChangedEvent] - OctoprintPrinterBedTemp changed from 54.96 °C to 55.16 °C
2020-02-16 18:56:22.274 [vent.ItemStateChangedEvent] - OctoprintPrinterBedTemp changed from 55.16 °C to 55.35 °C
2020-02-16 18:56:24.302 [vent.ItemStateChangedEvent] - OctoprintPrinterBedTemp changed from 55.35 °C to 55.51 °C
2020-02-16 18:56:32.296 [vent.ItemStateChangedEvent] - OctoprintPrinterBedTemp changed from 55.51 °C to 55.36 °C
2020-02-16 18:56:34.313 [vent.ItemStateChangedEvent] - OctoprintPrinterBedTemp changed from 55.36 °C to 55.25 °C
```

blalor, 2020-05-08:
There's a threshold that defaults to 0.1°C before emitting temperature changes. Is that sufficient?

CWempe, 2020-05-08:
I think 0.1 is to small for me. (see logs above).
I would prefer something like `0.3`.

Can I change that threshold or is it hard coded?

blalor, 2020-05-09:
It's in the plugin settings.  🔧 → MQTT (under *PLUGINS*) → Topics, then change the threshold under "Temperature messages".

CWempe, 2020-05-09:
Great.
This is exactly what I wanted.

Thanks.

closed 2020-05-09 by CWempe