OctoPrint/OctoPrint-MQTT #85: topic was octoprint and now is octoPrint??

y8s, 2020-05-12:
#### What were you doing?
using the auto update feature to update to latest version as of May 12, 2020. I use NodeRED to read octoprint's MQTT data and do things like turn on box lights and fan when temps reach a certain value. After updating, these automations were broken.

Ideally provide exact steps to follow in order to reproduce your problem: -->

1. ... allow octoprint to auto update plugin for MQTT
2. ... wait
3. ... warm up nozzle and wait for lights and fan to come on... 

#### What did you expect to happen?
I expected live temp data to be sent to node red via mqtt

#### What happened instead?
tool0 temp was stuck at 22.2

#### Version of OctoPrint
1.4.0

#### Version of the MQTT plugin
0.8.6 (upgraded from 0.8.2 today)

#### Used MQTT broker and its version
mosquitto built into raspberry pi

#### Link to octoprint.log

omitted.

#### Link to contents of Javascript console in the browser

omitted.

#### Screenshot(s)/video(s) showing the problem

omitted.

...For now my resolution is to rename the topic in the MQTT plugin settings... but bummer that it broke on upgrade.

jneilliii, 2020-05-13:
> ...For now my resolution is to rename the topic in the MQTT plugin settings... but bummer that it broke on upgrade.

So the "default" topic was different in version 0.8.2 versus 0.8.6?

y8s, 2020-05-13:
> > ...For now my resolution is to rename the topic in the MQTT plugin settings... but bummer that it broke on upgrade.
> 
> So the "default" topic was different in version 0.8.2 versus 0.8.6?

I'm not positive, and I'm not a software engineer, but overriding a setting like this shouldn't happen in a patch version upgrade.

jneilliii, 2020-05-13:
I agree, that's why I'm trying to figure it out. Do you remember what it was before and after the upgrade? That would help.

jneilliii, 2020-05-13:
Never mind. I found it.

@tedder : https://github.com/OctoPrint/OctoPrint-MQTT/blame/faac91c82ebb6c751980a4cc1b5312a56b5d7cbf/octoprint_mqtt/__init__.py#L107

y8s, 2020-05-14:
What's weird (or funny?) is that it wasn't changed to the offical name capitalization (capital O, capital P), but some kind of half-way chimera baby. 

Are the settings stored externally?

jneilliii, 2020-05-14:
The settings are stored in config.yaml like any other plugin, but if you don't ever change that setting it will not be in there. My assumption is that you were using the default and when @tedder pushed an update that default got changed. It wasn't changed from that previous default in the latest version but one of the one's in between the version you had previously and the latest, and the changed default has carried on since.

pcb1962, 2020-05-24:
It was changed in @tedder pull request:
push 'retain' flag to publish-time only #63
https://github.com/OctoPrint/OctoPrint-MQTT/commit/8d68d38083f7d3269dc5cb785c7f46bf9d3a9047

[image]

Been wondering for a few days why I was no longer seeing mqtt messages after update - haven't used octoprint or my printer for a year or so.

jneilliii, 2020-05-24:
Yeah, as already mentioned we'r know who changed it. Now the question is, do we change it back or just move forward? I vote  change it back, which will then break it again for people that haven't manually fixed their settings. @Kunsi @tedder you seem  silent over here in the plugin repo, are you guys maintainers or am I left making this decision on my own?

pcb1962, 2020-05-24:
I say change it back as the change was not announced, and the documentation says octoprint

y8s, 2020-05-24:
change it back. making it something random is weird. also prevent overriding existing settings if it's an upgrade.

jneilliii, 2020-05-24:
That's the hard part @y8s, it's not that it's overwriting a setting since default settings like this are not written to disk.

iceananas, 2020-12-11:
+1 for change it back. The new change makes it necessary to rename existing toolchains and doesnt provide any benefit.

Bra1nK, 2021-02-12:
Whichever gets done make sure the documentation matches the reality. New user here and it took me a while to figure out why nothing was being published to "octoprint/#"

jneilliii, 2022-08-07:
docs updated thanks to @andreasbrett

closed 2022-08-07 by jneilliii