OctoPrint/OctoPrint-MQTT #86: Redundant messages

smurfix, 2020-05-14:
#### What were you doing?

Printing on a Prusa+MMU and observing the MQTT messages.

#### What did you expect to happen?

No redundant messages.

#### What happened instead?

* An MMU-equipped Prusa doesn't have multiple hot ends, it has only one. Thus, please emit only one temperature event, not five identical ones for tool0…tool4. 

* I get a message for each Z change, which unfortunately includes retractions. Would it be possible to limit that to actual layer changes?

#### Version of OctoPrint

current

#### Version of the MQTT plugin

0.8.6

#### Screenshot(s)/video(s) showing the problem

```
octoPrint/temperature/tool4 {"_timestamp": 1589452785, "actual": 234.8, "target": 235.0}
octoPrint/temperature/tool3 {"_timestamp": 1589452785, "actual": 234.8, "target": 235.0}
octoPrint/temperature/tool2 {"_timestamp": 1589452785, "actual": 234.8, "target": 235.0}
octoPrint/temperature/tool1 {"_timestamp": 1589452785, "actual": 234.8, "target": 235.0}
octoPrint/temperature/tool0 {"_timestamp": 1589452785, "actual": 234.8, "target": 235.0}
octoPrint/event/ZChange {"new": 8.896, "_event": "ZChange", "old": 7.896, "_timestamp": 1589452786}
octoPrint/event/ZChange {"new": 7.896, "_event": "ZChange", "old": 8.896, "_timestamp": 1589452786}
octoPrint/event/ZChange {"new": 8.896, "_event": "ZChange", "old": 7.896, "_timestamp": 1589452787}
octoPrint/event/ZChange {"new": 7.896, "_event": "ZChange", "old": 8.896, "_timestamp": 1589452788}
octoPrint/event/ZChange {"new": 8.896, "_event": "ZChange", "old": 7.896, "_timestamp": 1589452788}
```

jneilliii, 2020-05-14:
This might be difficult as octoprint itself doesn't have a single nozzle, multiple extruder option in the printer profile itself does it? This plugin just reports what octoprint gives it, so maybe this should be more of a request to core octoprint, not this plugin.

tedder, 2020-05-14:
OctoPrint and octoslack detect this. I'll fill you in soon.

tedder, 2020-05-14:
Here it is. In the base OctoPrint, there's a setting in the UI for "shared nozzle", which handles 2-into-1 and MMU2-style nozzles:
[image]

[And then in the plugin, you simply need to test for that](https://github.com/fraschetti/Octoslack/blob/cdef534e8c0f12da563c283c38f168768b09eb0c/octoprint_Octoslack/__init__.py#L1516):
```
if shared_nozzle and nozzle_number and nozzle_number != "0":
  # only show the first nozzle if they are 'shared'
```

jneilliii, 2020-05-14:
Thanks @tedder didn't know that was there...