OctoPrint/OctoPrint-MQTT #95: octoprint mqtt do not respect time zone

Ivosch, 2021-02-09:
End time of job is not recalculated to raspberry system time zone.

cmroche, 2021-04-05:
Could you please clarify where you are seeing this, and what you are seeing? A screenshot would help.

I'm looking through the data sent to MQTT, and I'm not able to find any place an end time is reported. The only data sent are timestamps and relative times, which should not reflect any particular timezone at all. The application using this data is responsible for adjusting timezones accordingly.

Ivosch, 2021-04-12:
Hello, thanks for the reply. The time is in topics:
octoPrint/event/DisplayLayerProgress_layerChanged
or
octoPrint/event/DisplayLayerProgress_feedrateChanged

{"updateReason": "layerChanged", "totalLayer": "185", "currentLayer": "20", "lastLayerDuration": "0h:00m:27s", "lastLayerDurationInSeconds": 27, "averageLayerDuration": "0h:00m:19s", "averageLayerDurationInSeconds": 19, "currentHeight": "2.15", "currentHeightFormatted": "2.1", "totalHeight": "27.049999237060547", "totalHeightFormatted": "27.0", "feedrate": "1800.000", "feedrateG0": "-", "feedrateG1": "1800.000", "fanspeed": "100%", "progress": "36", "m73progress": "26", "printTimeLeft": "14m", "printTimeLeftInSeconds": 886, "printerState": "printing", "**estimatedEndTime**": "15:50", "estimatedChangedFilamentTime": "-", "changeFilamentTimeLeft": "-", "changeFilamentTimeLeftInSeconds": 0, "changeFilamentCount": 0, "_event": "DisplayLayerProgress_layerChanged", "_timestamp": 1618234530}

cmroche, 2021-04-12:
@Ivosch This attribute is coming from the event, it's not managed or introduced by this plugin, but further after looking to OctroPrint I cannot even find a source for it, not does my basic install have this data so I suspect it's being introduced from another plugin you are using. If this is the case, you would need to notify the plugin authors and ask them to modify the data field to use a timestamp so that other systems (MQTT) can correctly interpret the timezone. There is no universal way for the MQTT plugin to handle this otherwise.

jneilliii, 2021-04-12:
Those are custom events added by the DisplayLayerProgress plugin.

Ivosch, 2021-04-13:
ok, thanks for explanation, I'm closing this issue

closed 2021-04-13 by Ivosch