OctoPrint/OctoPrint-MQTT #97: Plugin Abandoned?

cmroche, 2021-03-16:
As per the policy on reporting plugins I would like to confirm that this plugin is still active and maintained as there are a growing number of unanswered issues, and a merge request that has no feedback for nearly a year.

@Kunsi @jneilliii  informs me that you are currently the lead on this plugin.

Thank you for your confirmation either way.

tedder, 2021-03-16:
> you are currently the lead

Who is the maintainer? (Is it me?)

Kunsi, 2021-03-17:
Well, i guess technically i'm the lead on this plugin, yes. However i keep having personal difficulties due to the ongoing situation, so I think the plugin is indeed "not actively maintained".

I plan to continue working on it if the situation normalizes again.

In the meantime I will try to atleast view and respond to pull requests.

cmroche, 2021-03-20:
Thanks @Kunsi .

I think that's a fair proposition, I'm ok to close this issue if you agree.

Kunsi, 2021-03-20:
@cmroche I think i'll keep it open and pinned until i got more time to work on it.

cociweb, 2021-11-21:
Hello In the meantime, could you please make an official new release (0.8.11) with the currently available master branch? The latest release has serious security issues for months (tls is not working the only insecure plain fallback is available)  - the current master branch has all the minimum and the proper fixes.  Minimal effort is required only for the administration. @jneilliii or @Kunsi or @foosel or @tedder. Thank's in advance.

DavidDeSloovere, 2021-12-19:
New version was just released 🎉
Time to close?

closed 2024-01-07 by jneilliii