OctoPrint/OctoPrint-MQTT #103: Wrong progress published

eloo, 2021-05-12:
#### What were you doing?

I'm sending some metrics over MQTT using this plugin to HomeAssistant. 
And it seems that the progress is calculated different form what i see in OctoPrint itself.

This can be seen that the published progress in mqtt is a good amount ahead of the progress in octoprint.

#### What did you expect to happen?
I would expect that the progress in the mqtt plugin matches the progress in octoprint.

#### What happened instead?
Octoprint progress: 42%
Mqtt plugin progress: 55%

See screenshots for more info.

#### Version of OctoPrint

Version: 1.6.1 
Plugins:
Cura Thumbnails
Dashboard
Detailed Progress
DisplayLayerProgress
Firmware Check
HomeAssistant Discovery
M73 Progress
MQTT
PrintTimeGenius Plugin
Printer Dialogs
Printer Notifications
Virtual Printer

#### Version of the MQTT plugin
0.8.10

#### Used MQTT broker and its version
Mosquitto

#### Screenshot(s)/video(s) showing the problem

[image]
[image]

Thanks

darkson95, 2021-11-11:
I have the same issue, but the progress published over MQTT is lower than the progress in OctoPrint:
[image]
[image]

The DiscordRemote plugin publishes the same value as the MQTT plugin. Can it be that the value in OctoPrint is wrong?

DavidDeSloovere, 2021-12-19:
Also seeing different values. Octoprint UI probably uses a different calculation.
From the UI, one uses gcode and yet another according to time elapsed vs time left. 
Only got my printer a week, so don't take my word for it.

My current print, pretty big, shows up as 20% in MQTT / HomeAssistant, but 30% in Octoprint UI.
20% does not match layers percentage, nor height percentage. I don't see where 20% comes from really.
[image]

DavidDeSloovere, 2021-12-19:
@darkson95 Do you also have a plugin that does progress calculation? 
I see the @eloo has `PrintTimeGenius Plugin` just like me. [Link to plugin](https://plugins.octoprint.org/plugins/PrintTimeGenius/)

The mqtt plugin uses the official octoprint progress api. 
But we might see different progress due to the fact that we have other progress plugins installed.
I don't know if the official api/plugin can automatically get the data from a 3rd party plugin, or will always use built-in values.
https://github.com/OctoPrint/OctoPrint-MQTT/blob/master/octoprint_mqtt/__init__.py#L19

jneilliii, 2021-12-19:
Yeah, Print Time Genius uses a time based percentage and overwrites the UI, whereas this plugin uses the built in progress percent which is file size based IIRC.

standardgbg, 2023-01-04:
HI,

Did you find any solution for this? The mttq plugin seems to have the possibility for other plugins to write data to it. But I cant figure out any way to get the PrintTimeGenius plugin to do this.

/D

montyfert, 2023-04-26:
hi is there any way to show the actual progress via mqtt then .
thanks 
martin

jneilliii, 2023-04-26:
It's possible to detect if PTG is installed and adjust the reported percentage in MQTT, but will take some coding work to integrate. I've done this before with SimplyPrint plugin and will see about possibly submitting a PR for it.

montyfert, 2023-04-26:
thanks
martin

jneilliii, 2023-04-27:
Added with the commit above, would be great if we could get some validation from anyone here interested to test it out before merging. Copy/paste the URL below into Plugin Manager > Get More > ...from URL and click Install. There may be very slight differences between what is reported on MQTT vs the OctoPrint UI because of rounding differences and the use of a 5 second repeating timer to collect the data, but it should be closer than what you've seen previously. 

```
https://github.com/OctoPrint/OctoPrint-MQTT/archive/refs/heads/feature/ptg_progress.zip
```

montyfert, 2023-04-27:
i'll give it a try this afternoonthanks 
martin

montyfert, 2023-04-27:
Hi @jneilliii 
Just did a small print and it was almost spot on only about 1% difference st most .
Thanks so much and keep up the good work 
It's much appreciated. 
Martin

jneilliii, 2023-04-27:
Yeah, that ~1% difference is about what I was seeing during my testing, probably because of where/when/how I'm doing the math calculations. It might not match up exactly with what the js implementation the PTG plugin is doing.

PR #134 (2024-01-07): PrintTimeGenius aware print progress
switch print progress to a 5 second repeated timer on print start to allow for use of time based progress percentage provided by Print Time Genius, resolves #103
closed 2024-01-07 by jneilliii