OctoPrint/OctoPrint-MQTT #105: Initial printTimeLeft update takes forever on long prints

nickrbogdanov, 2021-06-11:
When I start a long (8hr) print, I see a couple of octoPrint/progress/printing messages and one octoPrint/event/PrintStarted message on MQTT.  In the former messages, printTimeLeft is either 0 or null.  I don't see a valid printTimeLeft field on MQTT until the progress bar hits 1%, which takes several minutes on a long print job.  It would be nice if this field got populated and sent out over MQTT as soon as it is known to OctoPrint, so that anything that wants to display the estimated time (e.g. dashboards) could be updated sooner.

bglnelissen, 2022-02-24:
I have the same issue. Now I am using the layers for progress.

```
topic: "vyper/event/DisplayLayerProgress_heightChanged"
```

Which contains quite a lot of info as well as the printTimeLeft.

```
payload: "{"updateReason": "heightChanged", "totalLayer": "666", "currentLayer": "316", "lastLayerDuration": "0h:00m:22s", "lastLayerDurationInSeconds": 22, "averageLayerDuration": "0h:00m:22s", "averageLayerDurationInSeconds": 22, "currentHeight": "47.85", "currentHeightFormatted": "47.9", "totalHeight": "100.69999694824219", "totalHeightFormatted": "100.7", "feedrate": "7200", "feedrateG0": "7200", "feedrateG1": "2400", "fanspeed": "100%", "progress": "66", "m73progress": "", "printTimeLeft": "1h9m", "printTimeLeftInSeconds": 4173, "printerState": "printing", "estimatedEndTime": "13:23", "estimatedChangedFilamentTime": "-", "changeFilamentTimeLeft": "-", "changeFilamentTimeLeftInSeconds": 0, "changeFilamentCount": 0, "currentFilename": "model.gcode", "_event": "DisplayLayerProgress_heightChanged", "_timestamp": 1645701225}"

nickrbogdanov, 2022-10-25:
> I have the same issue. You can use DisplayLayerProgress_feedrateChanged or DisplayLayerProgress_heightChanged for progress.

So you're using [this additional plugin](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress) to generate those events as the corresponding Gcode commands are fed to the printer, then using OctoPrint-MQTT to send the events to clients?