OctoPrint/OctoPrint-MQTT #108: Could not connect to broker

Andoramb, 2021-09-24:
#### What were you doing?

Trying to connect to mosquitto container
1. Using docker (Octoprint in docker + HA + Mosquitto combo)
2. input IP, authentication (doublechecked)

#### What happened instead?

I've managed to catch in the log the following:
`octoprint.plugins.mqtt - INFO - Settings changed (broker_diff={'url': '192.168.1.1'}, lw_diff={}), reconnecting to broker
 octoprint.plugins.mqtt - DEBUG - Not connected, enqueuing message: octoprint/temperature/bed - {"actual": 26.56, "target": 0.0, "_timestamp": 1632477299}
octoprint.plugins.mqtt - INFO - Settings changed (broker_diff={'url': '192.168.1.100'}, lw_diff={}), reconnecting to broker
octoprint.plugins.mqtt - DEBUG - Not connected, enqueuing message: octoprint/temperature/bed - {"actual": 26.73, "target": 0.0, "_timestamp": 1632477305}
octoprint.plugins.mqtt - DEBUG - Not connected, enqueuing message: octoprint/temperature/tool0 - {"actual": 27.22, "target": 0.0, "_timestamp": 1632477317}`

Even after multiple enable/disable the MQTT plugin, changing the authentications, etc. It still just says _Not connected_ for some reason

#### Version of OctoPrint
OctoPrint 1.6.1 Python 3.8.10

#### Version of the MQTT plugin
MQTT (0.8.10)
HomeAssistant Discovery (3.3.0)
Tasmota-MQTT (0.3.8)

#### Used MQTT broker and its version
Docker

#### Link to octoprint.log
https://gist.github.com/Andoramb/700c13710297cfc4a5a4a31a49e32c48

#### Link to contents of Javascript console in the browser
Didn't see anything relevant

Andoramb, 2021-09-24:
There is something related to HomeAssistant discovery though:
`2021-09-21 18:03:27,817 - octoprint.plugin - ERROR - Error while calling plugin homeassistant
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_homeassistant/__init__.py", line 142, in on_after_startup
    self._generate_device_registration()
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_homeassistant/__init__.py", line 217, in _generate_device_registration
    self._generate_sensor(
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_homeassistant/__init__.py", line 505, in _generate_sensor
    self.mqtt_publish(topic, payload, allow_queueing=True)
TypeError: 'NoneType' object is not callable
2021-09-21 18:03:27,817 - octoprint.plugins.prettygcode - INFO - Pretty GCode.
2021-09-21 18:03:27,819 - octoprint.plugins.themeify - INFO - Themeify initialized.
2021-09-21 18:03:27,898 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'base_url': 'XXX', 'path': '/', 'query_string': 'l10n=en'}
2021-09-21 18:03:27,991 - octoprint.plugins.consolidatedtabs - INFO - OctoPi
2021-09-21 18:03:28,049 - octoprint.server.views - ERROR - Error while retrieving template data for plugin homeassistant, ignoring it
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/octoprint/server/views.py", line 1213, in fetch_template_data
    wizard_required = implementation.is_wizard_required()
  File "/usr/local/lib/python3.8/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_homeassistant/__init__.py", line 974, in is_wizard_required
    _retain = settings().get_boolean(
AttributeError: 'Settings' object has no attribute 'get_boolean'`

Not sure if that could be connected. Maybe there is a conflict with something else?
Thank you :)

jneilliii, 2021-09-27:
Your error is with the home assistant plugin you have installed. I don't use that one so don't know for sure what the issue might be.

cmroche, 2021-09-27:
@Andoramb I maintain the HA plugin, I'll take a look at this by the weekend. But, I'm really unable to read it. Could you repaste the error with the newlines for clarity please.

Andoramb, 2021-09-27:
@cmroche sorry for the layout, can you check the attached gist link?
I couldn't figure out how it may be related to the HA plugin though (as all was working fine before).
The rootcause seems to be related to the main Octo-MQTT plugin.
Something is not happening, as my log always says:
`octoprint.plugins.mqtt - INFO - Settings changed (broker_diff={'url': '192.1681.1.100'}, lw_diff={}), reconnecting to broker`
`octoprint.plugins.mqtt - DEBUG - Not connected, enqueuing message: octoPrint/temperature/soc - {"temperature": 44.0, "_timestamp": 1632747339}`

The login data should match (I didn't change anything). 
Running in docker (I can ping the mosquitto service IP).

Andoramb, 2021-09-27:
Nope... Mosquitto logs are never mentioning anything about "octoprint"
Took 2 steps back: 

- disabled all MQTT plugins inside Octoprint
- my docker compose is simplified, but looks like this:
```
octoprint:
  ports:
    - 8441:80  (I have other stuff here as well)
    - 1883:1883
mosquitto:
  network_mode: host
```

hmm.. still no connection to be seen anywhere