OctoPrint/OctoPrint-MQTT #111: Help configuring a Temp sensor

appleimperio, 2021-11-09:
Hi Im trying to get the Bed temp of my 3D printer to homebridge. Im getting this message in the homebridge logs after configuring the MQTThing temp sensor. I have use MQTT before but this is the first time I have to deal with a payload in the form of a JSON.

```
[11/9/2021, 2:24:23 AM] [Genesis Bed Temp] Received MQTT: genesis/temperature/bed = {"actual": 22.1, "target": 0.0, "_timestamp": 1636442663}
[11/9/2021, 2:24:24 AM] [Genesis Bed Temp] Ignoring invalid value [NaN] for Current Temperature - not a number
```

I don't know how to get the number from "actual". With the first line I have the values but the homebridge plugin wants only the temperature. I try to use the second line (which I'm pretty sure is wrong) with no success

```
   "topics": {
                "getCurrentTemperature": "genesis/temperature/bed",
                "apply": "return JSON.parse(message).actual;"
            }
```

how I can extract the value of the temp only?

Thanks

jneilliii, 2021-11-09:
I'm not sure of the extraction/parsing options available in your setup, but did you try `JSON.parse(message)["actual"]`?

appleimperio, 2021-11-09:
This is one of the example for configuring the plugin but I don't know how to adapt it to OctoMqtt

'''
{
    "accessory": "mqttthing",
    "type": "humiditySensor",
    "name": "TEST-Humidity_EXT",
    "url": "mqtt://Your IP address server",
    "username": "",
    "password": "",
    "topic": "tele/Spot_Erable/SENSOR",
    "caption": "",
    "sensorPropertyName": "SI7021",
    "topics": {
        "getCurrentRelativeHumidity": {
            "topic": "tele/Spot_Erable/SENSOR",
            "apply": "return JSON.parse(message)[\"SI7021\"].Humidity;"  
                },
        "getStatusFault": "",
        "getStatusTampered": "",
        "getStatusLowBattery": ""
            },
    "history": ""
}
'''
your statement gave a JSON error can't even try it

appleimperio, 2021-11-09:
@jneilliii thanks for your help I got it working with this string

```
"topics": {
    "getCurrentTemperature": {
        "topic": "genesis/temperature/bed",
        "apply": "return parseFloat(JSON.parse(message).actual);"
}
```

jneilliii, 2021-11-10:
Ah, that makes sense. It's expecting a number so you're having to force it.

closed 2021-11-10 by jneilliii