OctoPrint/OctoPrint-MQTT #117: MQTT subscribe on OctoPrint does not work.

molodeztom, 2022-03-13:
I am using Octoprint Plugin from here.
So I have a MQTT Broker on a separate raspberry "Controller".
I recieve MQTT Messages on the "Controller" as described for temperatures and so on. This works well!

Additionally I want to shutdown the OctoPrint Raspberry by sending an MQTT command to OctoPrint-MQTT plugin.

To do so I utilize the mqtt_subscribe helper by a modified mqtt_test.py file in the plugin folder.

I would expect the MQTT plugin to recieve the shutdown message and answer "Yay, recieved..." then shutdown the pi.

But since the Octoprint update to the latest python version it does not work anymore. 
I do not get the message "Yay, recieved…." nor does it shutdown.

#### Version of OctoPrint: V1.7.3 OctoPi 0.18.0 on Pi3 env.python.version: 3.7.3

#### Version of the MQTT plugin: 0.8.12

jneilliii, 2022-03-13:
This plugin only is for sending data out, it does not react to messages received from outside messages pushed to a topic IIRC. That was why I made the [MQTT Subscribe](https://plugins.octoprint.org/plugins/mqttsubscribe/) plugin.

molodeztom, 2022-03-13:
Thanks for an immediate answer. So the sample script does not work anymore? The Subscribe plugin does only support Octoprint REST API. Do you have a hint how to issue a raspbi shutdown command with this api?

molodeztom, 2022-03-13:
Ok I think I found it myself. 
REST API: /api/system/commands/core/shutdown  
REST Parameters  {  "confirm"."0" } 
or maybe
REST API: /api/system/commands/
REST Parameters  { "source"."core", "action"."shutdown", "confirm"."0" }

jneilliii, 2022-03-13:
I believe it's the first one, and I don't think it has any additional parameters necessary so you don't need the confirm I don't think.