OctoPrint/OctoPrint-MQTT #118: Attribute "port": "VIRTUAL" other topics

ilsicca, 2022-03-28:
octoprint.version: 1.7.3
env.os.bits: 32
env.plugins.pi_support.model: Raspberry Pi 3 Model B Plus Rev 1.3
MQTT (0.8.12)

Hi! I installed OctoPrint-MQTT Plugin and I have 3 printers conected in diferents serial ports. The topic octoprint/event/Connected has "port": "VIRTUAL".  The  othertopics lists hasn't  an attribute to identifiy a serial port, for example, the topic "octoprint/event/PrintStarted". How can I determine whose temperature it is? Who started printing?

Thanks!!

jneilliii, 2022-03-29:
change the base topic for each instance to something different, ie Printer1, Printer2, Printer3.

ilsicca, 2022-03-29:
Thanks @jneilliii ! 
I understand but if I have one Octoprint intances with 3 printers conected on my Raspberry Pi by USB?