OctoPrint/OctoPrint-MQTT #121: Missing dependency 'Six'

RoboMagus, 2022-05-27:
#### What were you doing?

After updating my docker Octoprint image to 1.8.0 I ran into issues relating to missing python module `six`, of which this plugins is one of ones that were throwing exceptions. 

See [this issue](https://github.com/OctoPrint/OctoPrint/issues/4528) in the octoprint repo.

Python module `six` is imported, but not listed as a dependency in `setup.py`.

#### What did you expect to happen?
No exceptions.

#### What happened instead?
See issue linked above.

#### Version of OctoPrint
1.8.0 (Docker)

#### Version of the MQTT plugin
Any

#### Used MQTT broker and its version
mosquitto

#### Link to octoprint.log
```
2022-05-25 18:27:20,666 - octoprint.startup - INFO - ******************************************************************************
2022-05-25 18:27:20,670 - octoprint.startup - INFO - Starting OctoPrint 1.8.0
2022-05-25 18:27:20,671 - octoprint.startup - INFO - ******************************************************************************
2022-05-25 18:27:21,259 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2022-05-25 18:27:21,314 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2022-05-25 18:27:21,314 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2022-05-25 18:27:21,314 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2022-05-25 18:27:21,505 - octoprint.startup - INFO - Blacklist processing done, adding 3 blacklisted plugin versions: roomtemp (any), GcodeEditor (>=0.1.1,<=0.2.8), gcodeleveling (>=0.1.0,<=0.1.1)
2022-05-25 18:27:21,519 - octoprint.plugin.core - INFO - Loading plugins from /usr/local/lib/python3.8/site-packages/octoprint/plugins, /octoprint/octoprint/plugins and installed plugin packages...
2022-05-25 18:27:22,029 - octoprint.plugin.core - INFO - Plugin Virtual Printer is disabled.
2022-05-25 18:27:23,718 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2022.3.1) is disabled.
2022-05-25 18:27:23,756 - octoprint.plugin.core - ERROR - Error loading plugin mqtt
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/core.py", line 1274, in _import_plugin
    module = _load_module(module_name, spec)
  File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/core.py", line 52, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/usr/local/lib/python3.8/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/usr/local/lib/python3.8/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 702, in _load
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_mqtt/__init__.py", line 5, in <module>
    import six
ModuleNotFoundError: No module named 'six'
```

Paraphraser, 2022-05-28:
Same problem. I'm running Octoprint in a Docker container. Fix:

```
$ docker exec octoprint pip install six
$ docker-compose up -d --force-recreate octoprint
```

chileocat, 2022-05-31:
I can confirm this problem with Octoprint 1.8.1 installed with pip. Plugin installed via plugin manager but doesn‘t work. After installing six manually with pip, it‘s fine.

jneilliii, 2022-05-31:
have patched setup.py and released new version since six was removed in OctoPrint 1.8.1.

PR #122 (2022-05-31): add six to requirements to resolve issue with OctoPrint 1.8.0+
resolves #121
closed 2022-05-31 by jneilliii