OctoPrint/OctoPrint-MQTT #129: [Feature Request] Publish current layer + total layers to mqtt

jsBlackBelt, 2022-12-19:
#### What were you doing?

Please add information of total print layers and current layer in the hass/progress topic, so it will be available in HA.

#### What did you expect to happen?
I expected to see total & current layer information in mqtt hass topic

#### What happened instead?
I could not find that information in mqtt hass topic

#### Version of OctoPrint
Version 1.8.6

#### Version of the MQTT plugin
0.8.13

#### Used MQTT broker and its version
mosquitto mqtt version 1.6.9

RoboMagus, 2023-01-17:
Having the plugin `DisplayLayerProgress` installed (and event messages enabled) there are added topics that include the information you're looking for.

chompy18, 2023-01-17:
I installed the `DisplayLayerProgress` plugin, but ho can I verify event messages are enabled?

RoboMagus, 2023-01-18:
> I installed the `DisplayLayerProgress` plugin, but how can I verify event messages are enabled?

Under the `MQTT` plugin settings, go to the `Topic` Tab, ther you'll find the `Activate event messages` checkbox.

Below is a screen snip showing the MQTT topics added. Though probably you'll have to start a print first for the topics to appear.

[image]

sefininio, 2023-01-18:
Got it, thanks. I'll check it out on the next print. Thanks!

sefininio, 2023-01-19:
OK, I do see those events on MQTT, but they are not discovered by HA, even though I have home assistant discovery plugin installed :(

RoboMagus, 2023-01-24:
Additional MQTT topics are indeed not auto discovered by HA, as the discovery plugin has no way to include that.

However there are wel documented ways to create MQTT [automations](https://www.home-assistant.io/docs/automation/trigger/#mqtt-trigger) and [sensors](https://www.home-assistant.io/integrations/sensor.mqtt/).

E.g. I've got a Current Layer sensor setup like so:

```yaml
mqtt:
  sensor:
    - name: 'Octoprint current layer'
      state_topic: 'octoPrint/event/DisplayLayerProgress_layerChanged'
      value_template: '{{ value_json.currentLayer }}'
```

closed 2024-01-07 by jneilliii