OctoPrint/OctoPrint-MQTT #130: No MQTT connected event message sent when Octoprint autoconnects to printer

rocket59, 2023-01-28:
#### What were you doing?

Looking to get a notification that my printer is connected.

Ideally provide exact steps to follow in order to reproduce your problem: -->

MQTT pugin installed with stock settings
Octoprint configured to autoconnect to printer on startup

#### What did you expect to happen?

Connected message received via MQTT broker

#### What happened instead?

No message received. If I terminate the printer connection and manually reconnect the printer the connected message will be sent

#### Version of OctoPrint

1.8.6

#### Version of the MQTT plugin

0.8.13

#### Used MQTT broker and its version

Mosquitto 2.0

jneilliii, 2023-01-28:
I suspect this is a race condition of some kind where the connection is established to printer prior to the plugin's code being loaded to react to the connection event.

rocket59, 2023-02-01:
Thanks - that makes sense. I imagine  that a lot of ppl would use the auto connect function so I'm a bit surprised to be the only one reporting this. Any thing I can do the speed up the plugin or delay the connection? Do plugins load in order in which they were installed? If so I could delete and re-add them with MQTT at the front.

Alternative is it possible to get additional server events like "PrinterStateChanged" sent via MQTT. There is an event payload called "operational" which could be used as a poxy for connected I guess.