OctoPrint/OctoPrint-MQTT #132: How to setup multiple printers?

Ali229, 2023-02-05:
#### What were you doing?

1. Add mqtt and discovery plugin to multiple printers
2. They get succesfully discovered in home assistant

#### What did you expect to happen?

Able to see data of each printer correctly/seperately

#### What happened instead?

The printers that get added after the first one just show the data of the the first one

#### Version of OctoPrint

1.8.6

#### Version of the MQTT plugin

0.8.13

#### Used MQTT broker and its version

Mosquitto broker
Current version: 6.1.3

#### Link to octoprint.log

https://pastebin.com/empZbpNQ

jneilliii, 2023-02-05:
I'm not sure how the discovery plugin works, but you could edit the topic in the MQTT settings to be able to identify each separately.

Ali229, 2023-02-05:
I have tried chasing the base topic attachment:
[image]

But then all printers become unavailable on home assistant attachment:
[image]

Ali229, 2023-02-05:
Never mind I had to change the topic on octoprint, delete the instance of mqtt on home assistant, and then change the discovery name on octoprint to make it rediscover on home assistant, very gimmicky but worked.

closed 2023-02-05 by Ali229