OctoPrint/OctoPrint-MQTT #135: MK3s "paused for user" due to collision

erichstuder, 2023-07-02:
#### What were you doing?

- Setup: MK3s with OctoPi ( and HomeAssistant which receives the MQTT messages)
  => Everything works fine so far.
- I was printing with my Prusa MK3s (I think the MMU2s is not relevant).
- The printer detected a collision 3 times in a row.
- The printer stopped and showed a message in the display that user action is necessary.
- No indication over MQTT that something was wrong (e.g. Print Status is still 'Printing')
- Althoug the printer sent information about what is going on to the OctoPi
  (e.g.  #531 2023-06-04 12:01:17,749 - Recv: echo:busy: paused for user)
  [serial.log](https://github.com/OctoPrint/OctoPrint-MQTT/files/11930095/serial.log)

Am I getting or doing something wrong?
By looking into the code my first impression is, that it could be, that this is just not implemented.
Is this something that could be added? (I might help)

Other people pointed out that this might be specific to Prusa:
https://community.octoprint.org/t/paused-for-user-support/52650
(Note: I mean the 3rd entry. My argumentation there was completely wrong otherwise.)

#### What did you expect to happen?

The plugin somehow reports that the printer is waiting for me.

#### What happened instead?

The plugin did not report the waiting/error via MQTT as far as I can see.

#### Version of OctoPrint

1.9.1

#### Version of the MQTT plugin

0.8.13

#### Used MQTT broker and its version

Mosquitto broker 6.2.1

#### Link to octoprint.log

[octoprint.log](https://github.com/OctoPrint/OctoPrint-MQTT/files/11930114/octoprint.log)

erichstuder, 2023-07-09:
I started trying out a possible implementation:
https://github.com/erichstuder/OctoPrint-MQTT/tree/paused_for_user

jneilliii, 2023-07-09:
After you've tested it for a bit feel free to send in a pull request.

closed 2024-01-07 by jneilliii